riffnshred/nhl-led-scoreboard #1: KeyError: 'totalGames'

jordanrvillarreal, 2019-10-25:
Running into an error.  

INFO (16:33:26): NHL Scoreboard - v0.1.0 (64x32)
http://statsapi.web.nhl.com/api/v1/schedule?teamId=None
Traceback (most recent call last):
  File "main.py", line 28, in <module>
    data = Data(config)
  File "/home/pi/nhl-led-scoreboard/data/data.py", line 30, in __init__
    self.refresh_fav_team_status()
  File "/home/pi/nhl-led-scoreboard/data/data.py", line 56, in refresh_fav_team_status
    self.fav_team_game_today = nhlparser.check_if_game(self.fav_team_id)
  File "/home/pi/nhl-led-scoreboard/data/nhl_api_parser.py", line 189, in check_if_game
    game = game_data["totalGames"]
KeyError: 'totalGames'

Trying to step through it on my own but not much success just yet.

riffnshred, 2019-10-26:
Fixed it. Update it and let me know if it works for you.

jordanrvillarreal, 2019-10-26:
Works!  Can't wait to see where you go with this!

closed 2019-10-26 by jordanrvillarreal