riffnshred/nhl-led-scoreboard #6: Option to delay updates

joshkay, 2019-11-23:
Just got this setup and the first thing I noticed was that the API updates faster than the feed I use to watch games.  Would you be open to a pull request to add this functionality or have you thought about a feature like this at all?

riffnshred, 2019-11-24:
It's in the plan in the next features. You will be able to set the delay your self in the config file. I tried it for a while and the only inconvenient is that the timing is always different. One day having a 1 min delay is perfect and the goal appears 30 sec after the feed, someday it will take 2+ min.

joshkay, 2019-11-24:
Ant chance you can commit what you have?  I'd rather not do something that you have already. Happy to try to improve it.

riffnshred, 2019-11-24:
I'm rewriting the Rendering section and part of the Data section so it's not really possible to commit this right now. 

If you really want to change yours a good place to start is, go to [renderer/main.py](https://github.com/riffnshred/nhl-led-scoreboard/blob/master/renderer/main.py) and add `time.sleep(60)` on line 142 just before the `self._draw_goal()`. 

I'm not adding any features to the current version (Unless someone make something that is functional) as I'm 100% focus on writing and testing V1.0 . I can assist you tho if you have questions and want to know where to start and how my code works. 

I'll create a Discord channel this today or in the next couple of days to make collaborations easier.

joshkay, 2019-11-24:
Okay no problem. I totally understand. I'll just hold off until your next commit. Looking forward to it! Thanks for all the work you've done so far it looks awesome.

riffnshred, 2019-11-24:
Thank you so much for the interest and kind words man. CHEERS !!! 😄 🍺