riffnshred/nhl-led-scoreboard #7: API + Website to control device

joshkay, 2019-11-26:
Ideas include:
- Changing saved options (fav team etc)
- Switching to other games

riffnshred, 2019-11-26:
External control would be great. Something like a basic web app you can access with you phone would be a must. I run my screen using Termius on Android right now so I dont have to get to my computer to turn it on etc...

joshkay, 2019-11-26:
Alright this gives me somewhere to start from. Let me know when you make a discord server!

riffnshred, 2019-11-26:
Will do !!!

riffnshred, 2019-12-06:
Link to the Discord server. Will add it to the README for the next update.

https://discord.gg/hZNntT

tandarra, 2019-12-19:
Have just stumbled across this project very excited have done MLB board about 2 years ago.
Discord link expired. 
Also is it possible to swap between games automatically like the MLB board does or only manually at this stage?

riffnshred, 2019-12-19:
> Have just stumbled across this project very excited have done MLB board about 2 years ago.
Thanks, glad your pumped !

> Discord link expired.
I'm currently at work, i'll send a new one tonight.

> Also is it possible to swap between games automatically like the MLB board does or only manually at this stage?
Not yet but this is where I'm going. You will have your list of preferred teams in the config file and those will be the teams that the scoreboard will focus as main scoreboard. you will have the option to cycle through the other games between periods. That's for V1.0 which I aim to release end of January. There are other functions coming too but nothing 100% solid for that release.

riffnshred, 2019-12-21:
Sorry for the delay. Here's a new link for the discord channel. https://discord.gg/CWa5CzK