riffnshred/nhl-led-scoreboard #8: Question regarding the update

her001, 2019-12-06:
Greetings,

First, great work on this project! I was super interested in what you have done and wanted to try it out myself, but also saw opportunity to make some improvements. Ordinarily, I'd just submit pull requests as I make changes, but as you seem to be working on a big update, I had this question instead:

Would it be possible to work on your update in the open, perhaps in a separate branch on this public repo? This way, I (and others) could contribute to the new version, or test out the new features. As it stands, changes to the master branch may be irrelevant or cause major merge conflicts when your private work is ready to be released.

Thanks!

joshkay, 2019-12-06:
Hey he just posted his latest updates and added a discord link.  Join the discord and come talk!

her001, 2019-12-06:
Ah yes, now I see the [Dev](https://github.com/riffnshred/nhl-led-scoreboard/tree/Dev) branch. Thanks @joshkay!

closed 2019-12-06 by her001

her001, 2019-12-06:
@joshkay If you don't mind me asking, where is the Discord?

riffnshred, 2019-12-06:
Hi @her001 here it is. I just made the chanel last night. https://discord.gg/hZNntT 

As for working on the V1.0 it wont be possible for now. I appreciate the hype everybody has about working on this with me, but I want to lay the foundation of this project (what I want to see it become) first before I "send this kid out in the wild" if you know what I mean xD. 

What's in the Dev branch right now is what the current state i'm at. I the discussion my and Josh had on discord last night you'll see the layout for now. I have a full plan on papers which I'll post in the Wiki in the next week or so.