riffnshred/nhl-led-scoreboard #9: Need housing information [hardware]

ZLevine, 2019-12-10:
First of all, awesome job on this project! I was thinking about covering this project in a [howchoo](https://howchoo.com/) guide and video, so I'm in the requirements-gathering phase.

I know the MLB-LED-Scoreboard guys covered hardware in [their Wiki](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/wiki), but this doesn't cover the actual housing for the screen; did you 3D print yours? If so, uploading the source files (STL at a minimum) to the repo would be a great addition.

Would love additional info on the housing-portion of the hardware setup. Thanks!

desigi, 2019-12-10:
I had mine 3d printed. I modified one that I found on thingiverse. It's not perfect as the spacing of the holes along the width of the board is off by maybe 3-4mm. I haven't had it reprinted because it cost quite a bit to print the first one. Otherwise it works quite well. I added a mounting point for the Raspberry Pi also. Hope this helps. 

https://tinyurl.com/s7h4xvv

As a reference, the Raspberry Pi takes M2.5 bolts and the LED board takes M3 bolts.

riffnshred, 2019-12-10:
Nice Case @desigi . It's really thick, that's probably why it cost a lot, but I like the back. 

@ZLevine Thanks for the good words and thanks for doing this man. I was planning on making one, but I'm focused on finishing V1.0. I'll send you the STL file of mine tonight (at work atm). I did'n make the model, I found it on an Arduino project which I can't find anymore. The only issue is that it's far from perfect. The Pi is just loose in the enclosure. The biggest selling point was that the screen actually slide in the enclosure. It makes it look a lot better I find. The solution I have is replace the back of the case for a plexiglass plate and just glue the bolts and screw the Pi on it. That could be great if I add a led strip in the case and when there is a goal, just have the strip flash Red along with the goal animation.

Someone on reddit contacted me about my case and I gave him my ideas and he told me his and he said he's going to give it a shot and make one with our specs (which If he does will be fantastic). I'll post the specs of the case I envision for it in the Wiki page and if someone make it and it's tested and looks great, I definitely will add that case to the project.

riffnshred, 2019-12-10:
Here are the stl files I used. I don't even use the back part anymore as it's kinda hard to take off. 

[3D Printed Case.zip](https://github.com/riffnshred/nhl-led-scoreboard/files/3947392/3D.Printed.Case.zip)

ZLevine, 2019-12-11:
Thanks so much! I’ll try this one out. Definitely post the other one too whenever he sends it your way.

Also, the LED strip is an interesting idea :) I’ll keep that in mind.

ZLevine, 2019-12-21:
So I accidentally ordered the 5mm pitch panel so neither of these housings worked, but [this one did](https://www.thingiverse.com/thing:3779303)! So it's important to note that there's a difference between 3mm (P3), 4mm (P4), 5mm (P5), 6mm (P6), etc. since the housings are physically a different size.

closed 2019-12-21 by ZLevine

riffnshred, 2019-12-21:
Sorry I forgot to specify that. Yes my screen is a P3 (3mm spacing between each pixel). Love the case you found.