riffnshred/nhl-led-scoreboard #12: Game clock should actually count down

ZLevine, 2019-12-21:
Currently, live game data is retrieved and the current game clock value is statically displayed.

Instead, the game clock should count down as soon as data is retrieved, so you can see how much time is left in the period. This will be a relatively trivial coding change, but will make the clock far more dynamic. :)

riffnshred, 2019-12-21:
Its a nice feature, but to me its fake data. Unless there is an actual way to have it catch when the game clock start/pause/finish in under 10 sec I feel like it mess up the user experience.

That being said, you are not the only one requesting it. If its well wrote as an option that the user may set for him self, I don't have an issue accepting a pull request. 

I'd say wait for V1.0 as the code completely change and It coded more like the MLB one.

riffnshred, 2019-12-21:
A clue on where to start is in the Live overview of a game like this https://statsapi.web.nhl.com/api/v1/game/2019020561/feed/live

There is a timestamp in the Metadata that tell at what time the data was refreshed.

jeepdrvr71, 2020-01-31:
I think a good addition on no game days would be to display the standings in the division of your favorite team. Or a clock and weather from your local area.

riffnshred, 2020-01-31:
That is a whole other issue and not relevant to this one. But dont worrie, its coming.

https://streamable.com/lrv3e

https://streamable.com/bfh5g

https://streamable.com/gw8o6

Not everything in these clip are set in stone but its a good idea of what its going to be

jeepdrvr71, 2020-02-01:
I know it wasn’t relevant to the thread. The videos you posted are amazing. Any idea of when they will be available? I love the standings instead of No Game Today.

riffnshred, 2020-02-01:
The current situation is, next weekend I upload a beta version on the dev branch of V1 for testing If you want to join in, just join the Discord channel and I'll be posting the details there. https://discord.gg/CWa5CzK

I don't have an exact date for the official release at the moment but I'm confident that it will be by the end of February.