riffnshred/nhl-led-scoreboard #13: Period recap

blaineA, 2019-12-31:
Being able to see who scored at the end of each period and would be a nice touch. Something like a recap screen that last about 5-10 minutes in between periods. Just a thought I was kicking around for a few days, could be a little to much/ unnecessary for the project.

riffnshred, 2019-12-31:
It's a great idea!

I'm currently developing a routine system that shows and rotate "Boards" of information. The first boards available will be a list of all the games of the day and the Standings. More boards will be added over time and the User will be able to select what to show before the game, during intermission, and after the game. I thought it would be a good system to make it easier to integrate new features into the project in the future.

ruchoff, 2020-03-01:
Could also have it show team stats like SOG, power plays, PP%, PIM, face-off win %, blocked shots, takeaways, giveaways, and hits.

riffnshred, 2020-03-01:
Great idea. Love it

ruchoff, 2020-03-01:
Possible mockups
[image]
[image]