riffnshred/nhl-led-scoreboard #14: Black box for scoreboard

msroth, 2020-01-07:
I like be black box the scoreboard in the photo is in.  Did you make it or buy it?  Care to share the details?

riffnshred, 2020-01-07:
It`s a 3D printed case I found on another project. the issue with it is that it`s not made to have a Raspberry pi in it. Some are working on making a proper one

mrdink, 2020-01-07:
A great tutorial on how to make one is located here https://howchoo.com/g/otvjnwy4mji/diy-raspberry-pi-nhl-scoreboard-led-panel

I had to modify the model to fit my screen but it worked out great.

[image]

joshkay, 2020-01-07:
What size screen do you have?

mrdink, 2020-01-07:
> What size screen do you have?

It’s the same size but it doesn’t have the frame on the back of it.

https://www.amazon.com/dp/B07S2VZ13G?ref=ppx_pop_mob_ap_share

riffnshred, 2020-01-08:
Oooh kk you got the flexible version. Glad it fit in the case after all

closed 2020-01-08 by riffnshred