riffnshred/nhl-led-scoreboard #15: key errors

mgaerttner, 2020-01-13:
Greetings,
I'm running the latest version of raspbian with python 2.7.16-1. I was getting key errors on a fresh install when trying to start the scoreboard app. In short, it seems like it didn't want to pick up any of the settings/variables in the json config files (was receiving key errors for end_of_day, debug, fav_team_id). I made the following modifications to get things to run.

data/scoreboard_config.py:

from:

> from utils import get_file
> import json
> import os
>
>
>class ScoreboardConfig:
>    def __init__(self, filename_base, args):
>        json = self.__get_config(filename_base)
>        # Misc config options
>        self.end_of_day = json["end_of_day"]
>        self.debug = json["debug"]
> 

to:

> 
> from utils import get_file
> import json
> import os
>
> with open('config.json', 'r') as f:
>    config = json.load(f)
>
> class ScoreboardConfig:
>    def __init__(self, filename_base, args):
>        #json = self.__get_config(filename_base)
>        json = config
>        # Misc config options
>        self.end_of_day = json["end_of_day"]
>        self.debug = json["debug"]
> 

I'm definitely not a python programmer and I'm sure there's a much cleaner way to do it. I just wanted to post in case it helps anyone out.

riffnshred, 2020-01-13:
Thanks !!! 

I'll test it later tonight with a fresh install!

mgaerttner, 2020-01-15:
Hey is there a contrib location that files can be uploaded to for your project? I created a simple bash startup script and a systemd service file that can be used with cron for automating starting/stopping. I can upload them if you'd like.

Thanks!

riffnshred, 2020-01-17:
Not for now. I don't take any contribution (there are a few exceptions depending on what it is) until I release V1.0. The reason is the whole code change. Soon after the release, I'll organize the repo to make it easy to work with contributors along with what my vision is for the future of this project and what are the priorities.

riffnshred, 2020-03-02:
We are recommending using Supervisor now instead. If you have followed the Howchoo guide and or install a bash script to run the board automatically, undo that process or start with a fresh install of Rasbian Lite than fallow the instruction to install supervisor in the readme.

closed 2020-03-02 by riffnshred

mgaerttner, 2020-03-08:
Supervisor install method works great. Software is looking good. Thanks to you and everyone else for your efforts!

riffnshred, 2020-03-08:
Welcome bud ! Enjoy!