riffnshred/nhl-led-scoreboard #16: Multiple favorite teams?

xxxDrewedxxx, 2020-01-17:
I come from a house with a Blues fan (roll eyes) a Bruins fan (ugh) and a Wild fan (me.)  Is there some way to a momentary switch that would allow the program to move to the next "favorite" team. I.E. It is displaying the Wild game <push button> now displays the Blues, < push again> displays Bruins< push again> back to the Wild.  
Don't know if it is even possible.   
Also is there a way to use a monitor and not the LED matrix?

riffnshred, 2020-01-17:
Hah I feel your pain. I'm a Habs fan surrounded by Ottawa fans. 

It will be possible in a way on the next release. I made a board system where the user chooses what to show when the scoreboard is on a specific state (Off days, Pre-game, intermission, etc...) and one of the first boards that will be available at lunch is a Score Ticker. The one option that's programmed for that board is to rather show all the games or just the preferred games. The only issue, for now, is that the score animation won't be available for that board as it just goes through a list of games and shows the current info. 

So with that, you have 2 options:
1. set your Wild as the first team in the preferred team's list, the days they play, their game will be set as the main event of the day. Then set the Score ticker to show the preferred teams only so that before, during intermission, and after the game, it will show the other games score. 

2. If option #1 causes problems, you just set the scoreboard to show the Boards and ignore the main event. You put the Score Ticker as the only board and adjust the refresh rate to your taste and voila !!!. The only issue as mentioned is that there is no animation when there is a goal.

riffnshred, 2020-01-17:
Ill keep that issue open as I might add options surrounding this idea. I already programmed some part with the idea of showing the next favorite team's game after the first one is done (for those days where one is playing in the afternoon of the other is still ongoing after the main one is done etc...)

xxxDrewedxxx, 2020-01-18:
Sounds good.  My led matrix and other goodies are on their way to me.  I’ll be making this as soon as it arrives

WashCaps33, 2021-03-15:
> Hah I feel your pain. I'm a Habs fan surrounded by Ottawa fans.
> 
> It will be possible in a way on the next release. I made a board system where the user chooses what to show when the scoreboard is on a specific state (Off days, Pre-game, intermission, etc...) and one of the first boards that will be available at lunch is a Score Ticker. The one option that's programmed for that board is to rather show all the games or just the preferred games. The only issue, for now, is that the score animation won't be available for that board as it just goes through a list of games and shows the current info.
> 
> So with that, you have 2 options:
> 
> 1. set your Wild as the first team in the preferred team's list, the days they play, their game will be set as the main event of the day. Then set the Score ticker to show the preferred teams only so that before, during intermission, and after the game, it will show the other games score.
> 2. If option #1 causes problems, you just set the scoreboard to show the Boards and ignore the main event. You put the Score Ticker as the only board and adjust the refresh rate to your taste and voila !!!. The only issue as mentioned is that there is no animation when there is a goal.
> 
> As for the monitor it's not programmed for that sadly.

How can this be done?  I see all these things people are doing, multiple teams, score animations, etc.  Is there documentation anywhere to guide a user to use these modifications?

riffnshred, 2021-03-15:
All the documentation for this is in the wiki section. To change what to display at which state, open up the nhl_setup to configure your scoreboard and for each state put whatever boards you want.

closed 2021-03-15 by riffnshred