riffnshred/nhl-led-scoreboard #17: "

PHI34Halladay, 2020-01-20:
Hey guys,
I just installed the NHL Scoreboard. I'm using the MLB Scoreboard since early April 2019 and just found out that there is a NHL version :)

My problem now is that whatever I do, I only get the "no game today" info. I tried changing the favorite team id to a team playing today in the config.json and by using the --fav-team flag without any results.
It's January 20th today and Detroit is playing but using the ID "17" still gives me the same info:

INFO (11:04:28): NHL Scoreboard - v0.1.0 (64x32)
INFO (11:04:29): Off day State
INFO (11:04:29): ping_day_off

Is there an option somewhere to check the correct date?

Please help me out here.

Regards,
Patrick

riffnshred, 2020-01-20:
How is it now?

PHI34Halladay, 2020-01-20:
Works like a Charme :)

riffnshred, 2020-01-20:
It's because the API update to that new day at Noon everyday ;). On the new version that's coming out, you can set at what time you want the board to update to the new day. 

Cheers

closed 2020-01-20 by riffnshred