riffnshred/nhl-led-scoreboard #18: 132x64?

PHI34Halladay, 2020-01-21:
Since I have four LED panels connected I was wondering if there are any plans for a 132x64 version.

I found the file to readjust the team logos in the renderer folder but didn't find with the file where I can readjust the game info.

riffnshred, 2020-01-21:
Its not in the current plan but once the major features for 64x32 are done I'll be open to PR for other ratio, but my priority will be the single boards since its more accessible and affordable.

riffnshred, 2020-01-22:
That's great actually. My goal is to finish the main features by summer and then expend options for the user. As I release those features you can adapt them to you bigger screen. 

I got to tell you tho, I made a board system to allow the user to have what ever they want in between games and periods. That can be a clock, weather, news, gif animations, whatever is available to them. I'll have to add a function that tell the user that certain boards are not available for their resolution. So unless you adapt those different boards, some wont be available to that resolution. I don't expect all boards to work on all resolutions ( I don't think a news ticker would look great on a 32x32) but just a headup that if you want some of those features you gonna have a lot of work.

I strongly suggest you play around and create stuff with the rpi-rgb-matrix repo so you understand how it works, that will save you a lot of time. You can do a lot with the Pillow library but there are some limitations.

PHI34Halladay, 2020-01-22:
Sounds good. I will give it a look the next days.

I also joined the Discord server, so see you over there :)

PHI34Halladay, 2020-02-05:
**Update on the 128x64 Version**
[image]

riffnshred, 2020-02-05:
Sweet. That's a lot a real estate.

jeepdrvr71, 2020-03-01:
I like seeing the full logos that size!

ThePapatech, 2020-03-08:
That is fantastic...

purist69, 2020-03-26:
any chance you could provide your code and configurations required to build this at 128x64?