riffnshred/nhl-led-scoreboard #49: Main.py immediately crashes

desigi, 2020-02-17:
I'm running the latest `beta` branch and main.py immediately crashes upon running it. 

This is the error I receive:
```
  File "src/main.py", line 47, in <module>
    run()
  File "src/main.py", line 43, in run
    MainRenderer(matrix, data).render()
  File "/home/pi/nhl-board-dev/src/renderer/main.py", line 38, in render
    self.__render_offday()
  File "/home/pi/nhl-board-dev/src/renderer/main.py", line 59, in __render_offday
    self.boards._off_day(self.data, self.matrix)
  File "/home/pi/nhl-board-dev/src/boards/boards.py", line 26, in _off_day
    board(data, matrix)
  File "/home/pi/nhl-board-dev/src/boards/boards.py", line 65, in scoreticker
    Scoreticker(data, matrix).render()
  File "/home/pi/nhl-board-dev/src/boards/scoreticker.py", line 28, in render
    ScoreboardRenderer(self.data, self.matrix, Scoreboard(self.games[self.index], self.data.teams_info)).render()
  File "/home/pi/nhl-board-dev/src/renderer/scoreboard.py", line 37, in render
    self.draw_live()
  File "/home/pi/nhl-board-dev/src/renderer/scoreboard.py", line 83, in draw_live
    self.draw_power_play()
  File "/home/pi/nhl-board-dev/src/renderer/scoreboard.py", line 131, in draw_power_play
    colors[str(away_number_skaters)])
KeyError: '0'
```

falkyre, 2020-02-17:
Any errors when running the install.sh?

desigi, 2020-02-17:
No. The only that might qualify as an error was:
```
Cannot uninstall 'rgbmatrix'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.
```

riffnshred, 2020-02-17:
I got it. My bad, I miss wrote the condition for powerplay. rerun your  install.sh and it should be fixed. I'll keep an eye out see if it does it again. thanks

desigi, 2020-02-17:
All is working again. Thanks!

closed 2020-02-17 by desigi