riffnshred/nhl-led-scoreboard #50: Suggestion to integrate Supervisor to monitor/restart the process

desigi, 2020-02-17:
I used to use `nohup` and `&` to detach the process from my terminal and run it in the background, but I would notice that sometimes the process would just fail and I'd have to log back in and run the process again.

I would suggest using something like Supervisor to monitor the process and restart it automatically if it stops. This is the config file that I am currently running:
```
[program:nhl]
command=python3 /home/pi/nhl-board-dev/src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=30 --led-slowdown-gpio=2
autostart=true
autorestart=true
stderr_logfile=/var/log/nhl.err.log
stdout_logfile=/var/log/nhl.out.log
```

falkyre, 2020-02-17:
That's what I'm doing. Others use screen or tmux but there's no restart. You can do the same with systemd if you want.

falkyre, 2020-02-17:
And bonus is you can have a web interface to your processes with supervisor

riffnshred, 2020-02-21:
Will definitely test that out this weekend and maybe integrate that for V1 release

riffnshred, 2020-03-02:
this is now one of our recommended solution as it allows controle using your phone and to switch to the MLB easily

closed 2020-03-02 by riffnshred