riffnshred/nhl-led-scoreboard #55: Additional "off day" option - clock

harmgsn, 2020-02-25:
Enhancement Req:  Add an option for off-day that shows a simple clock (12 or 24h) and maybe the upcoming dates/times for favorite teams all on one display...  nothing scrolling, just a simple display of a clock and date/times:

Example:

-----------------------------------------
|                     Mon 2/24                      |
|                         21:48                         |
|DAL 2/29 @ 18:00  CGY 3/1 @19:00 |
-----------------------------------------

(please forgive the janky ASCII art...  an artist I am not)

riffnshred, 2020-02-25:
Agreed and also that could be the fallback display, in case something goes wrong in the process loading the api data

jeepdrvr71, 2020-02-27:
I still think having the current playoff standings scroll is the best option on non-game days.

riffnshred, 2020-02-27:
Check out the details in the beta branch, you have options now to choose what to show at different state (offday, scheduled, intermission and post-game)

jeepdrvr71, 2020-02-27:
riffnshread I looked in the beta section and saw the new stuff. Is there a different version of the software that I need to download or is all this stuff in the general release that I downloaded a few weeks back?

riffnshred, 2020-02-27:
If you still see the "no game day" you are on V0.1 released in november. The stuff thats in the beta branch is what we have been testing for the last week and a half now. But its missing the clock and the fallback incase the screen has trouble loading the data. Id say wait till saturday when I officialy release it because you are going to need to reinstall it then anyway.

riffnshred, 2020-03-02:
the clock board is now the fall back option incase there is no data

closed 2020-03-02 by riffnshred