riffnshred/nhl-led-scoreboard #60: Getting Error

jeepdrvr71, 2020-02-29:
I'm installing the Beta and got this error. Any suggestions

pi@scoreboard:~nhl-led-scoreboard $ sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2

Traceback (most recent call last):
    File "src/main.py", line 9,  in <module>
    from dimmer import Dimmer
    File "/hom/pi/nhl-led-scoreboard/src/dimmer.py", line 1, in <module>
    import ephem
ModuleNotFoundError: No module named 'ephem'

riffnshred, 2020-03-01:
Did you install PIP 3 ? 

Also forget the Beta, the Official version is out now. Check out the documentation to install PIP3 (top of page) and then, pass from the beta to the Master version do these commands

`git reset --hard`
`git checkout master`
`sudo git pull`
`sudo chmod +x scripts/install.sh`
`sudo ./scripts/install.sh`

jeepdrvr71, 2020-03-01:
I installed PIP3 but still getting the same error. Should I wipe it out and start over?

jeepdrvr71, 2020-03-01:
Got it to final work. The only issue I am having now is loading the supervisor. I get the following

E: Could not open lock file /var/lib.dpkg/lock-frontend - open (13: Permission denied)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), are you root?

I made sure I was in the root. And still got this error.

jeepdrvr71, 2020-03-01:
I did this as a work around. Thanks Tyler for this script!!

nano ~/start-scoreboard.sh

#!/bin/bash
cd /home/pi/nhl-led-scoreboard/src
n=0
until [ $n -ge 10 ]
do
   python3 main.py --led-gpio-mapping=adafruit-hat --led-brightness=60 --led-slowdown-gpio=2 && break
   # Uncomment and use this line instead if you did the anti-flicker mod:
   #python3 main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2 && break
   n=$[$n+1]
   sleep 10
done

chmod +x ~/start-scoreboard.sh

sudo crontab -e

@reboot /home/pi/start-scoreboard.sh > /home/pi/cron.log 2>&1

sudo reboot

riffnshred, 2020-03-01:
that's an other way to do it, although you wont have the web interface. If it`s not important than BAM you are good. but If you ever want to install the MLB scoreboard and switch easily between both leagues you will need supervisor. at what point did this error show up ?

falkyre, 2020-03-01:
The error you are getting on install is because you don't have permissions to the file as the directory/file is owned by the root user.   The command in the readme is missing the sudo before it, so installing supervisor should be done:

**sudo apt-get install supervisor**

The error message is telling you that you aren't the root user so using sudo elevates your normal user (typically name pi) to the root user.  Try the sudo apt-get install supervisor and see if it installs.

The one drawback to the start-scoreboard.sh method is that if you get 10 failures of the script, it won't restart it after the 10th try.  With supervisor, you have unlimited retries.  But it all comes down to what you want to use.

jeepdrvr71, 2020-03-01:
I think I had a brainfart and didn't use sudo in the command. I got the supervisor up and running. I'm assuming that I should delete my previous start up job to match what the current build will work with.

I really like the idea of switching between baseball and hockey. My goal would be to do something similar for the English Premier League.

Thanks again falkyre and riffnshred for your help. This group is truly awesome. I'll post some pics of my build. I put it in a barnwood case that I built. Really looks great with all of the added features now.

falkyre, 2020-03-01:
Yah use one or the other not both.  If you use the start-scoreboard.sh and supervisor, you'll get two python process launched as the start-scoreboard.sh will just luanch another one up to 10 times and supervisor will just keep restarting as the process it controls fails.   You'll end up with a mess on your display.

jeepdrvr71, 2020-03-01:
Do you know how the best way to remove the other process? I would like to use the way it's detailed in this group so I can have it the same as everyone else.

falkyre, 2020-03-01:
The start_scoreboard.sh?  You need to edit your crontab.

`sudo crontab -e`

Remove this line

`@reboot /home/pi/start-scoreboard.sh > /home/pi/cron.log 2>&1`

Save the file.  

Press CTRL and X
press y
press enter

jeepdrvr71, 2020-03-01:
Did that and still don't get the display to run automatically. There is still an old log file being generated that is trying to run the old script

falkyre, 2020-03-01:
What is in your supervisor setup?

jeepdrvr71, 2020-03-01:
[program:scoreboard]
command=sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=65 --led-slowdown-gpio=2
directory=/home/pi/nhl-led-scoreboard
autostart=true
autorestart=true

falkyre, 2020-03-01:
ok, and you installed  supervisor using the sudo apt-get install supervisor?  Is supervisor running?  What do you get if you do sudo supervisorctl? Also, what's the name of the log file that's referencing the old script?

falkyre, 2020-03-01:
This is my supervisor running:

```dietpi@scoreboard:~/source/sbio$ sudo supervisorctl
sbdev                            STOPPED   Not started
sbio                             STOPPED   Not started
scoreboard                       STOPPED   Mar 01 10:44 AM
supervisor_alert                 RUNNING   pid 30915, uptime 4:38:29
supervisor>
```

jeepdrvr71, 2020-03-01:
I did install supervisor and I can get to the webpage. On the webpage I see that the state is running and the description is pid 3548, uptime 0:00:03

When I do sudo supervisorctl. I get the following
scoreboard                STARTING
supervisor>

The log file I was referring to is cron.log

falkyre, 2020-03-01:
What's crontab -l show

jeepdrvr71, 2020-03-01:
no crontab for pi

falkyre, 2020-03-01:
Ok. sudo crontab -l. That will show the root user crontab. What's it show?

jeepdrvr71, 2020-03-01:
It shows a file with a bunch of remarks. 

# Edit this file to introduce tasks to be run by cron.
#
# Each task to run has to be defined through a single line
# indicating with different fields when the task will be run
# and what command to run for the task
#
# To define the time you can provide concrete values for
# minute (m), hour (h), day of month (dom), month (mon),
# and day of week (dow) or use '*' in these fields (for 'any').
#
# Notice that tasks will be started based on the cron's system
# daemon's notion of time and timezones.
#
# Output of the crontab jobs (including errors) is sent through
# email to the user the crontab file belongs to (unless redirected).
#
# For example, you can run a backup of all your user accounts
# at 5 a.m every week with:
# 0 5 * * 1 tar -zcf /var/backups/home.tgz /home/
#
# For more information see the manual pages of crontab(5) and cron(8)
#
# m h  dom mon dow   command

falkyre, 2020-03-01:
Ok so shouldn't have a crontab now. Have you rebooted since changing or adding supervisor?

jeepdrvr71, 2020-03-01:
Yes several times

falkyre, 2020-03-01:
Can you paste what the cron.log file looks like? And the supervisord.log?

falkyre, 2020-03-01:
Should be in /var/log/supervisor I think.

jeepdrvr71, 2020-03-01:
This is the cron.log file. It appears to be referencing the old method I was using to start the scoreboard. I thought I deleted everything out of there.

/bin/sh: 1: /home/pi/start-scoreboard.sh: not found

This is the supervisord.log file

2020-03-01 08:56:56,668 CRIT Supervisor is running as root.  Privileges were not dropped because no user is specified in the config file.  If you intend to run as root, you can set user=root in the config file to avoid this message.
2020-03-01 08:56:56,686 WARN No file matches via include "/etc/supervisor/conf.d/*.conf"
2020-03-01 08:56:58,938 INFO RPC interface 'supervisor' initialized
2020-03-01 08:56:59,028 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2020-03-01 08:56:59,068 INFO supervisord started with pid 785
2020-03-01 09:19:19,108 CRIT Supervisor is running as root.  Privileges were not dropped because no user is specified in the config file.  If you intend to run as root, you can set user=root in the config file to avoid this message.
2020-03-01 09:19:19,114 INFO Included extra file "/etc/supervisor/conf.d/scoreboard.conf" during parsing
2020-03-01 09:19:19,293 INFO RPC interface 'supervisor' initialized
2020-03-01 09:19:19,296 CRIT Server 'inet_http_server' running without any HTTP authentication checking
2020-03-01 09:19:19,301 INFO RPC interface 'supervisor' initialized
2020-03-01 09:19:19,304 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2020-03-01 09:19:19,308 INFO supervisord started with pid 417
2020-03-01 09:19:20,317 INFO spawnerr: can't find command '[sudo'
2020-03-01 09:19:21,338 INFO spawnerr: can't find command '[sudo'
2020-03-01 09:19:23,368 INFO spawnerr: can't find command '[sudo'
2020-03-01 09:19:26,452 INFO spawnerr: can't find command '[sudo'
2020-03-01 09:19:26,483 INFO gave up: scoreboard entered FATAL state, too many start retries too quickly
2020-03-01 09:20:33,881 ERRO 192.168.1.36:53611 - - [01/Mar/2020:14:20:33 -0500] "GET /?message=ERROR%3A%20Process%20scoreboard%3A%20no%20such%20file HTTP/1.1" 200 2645
2020-03-01 09:24:23,585 ERRO 192.168.1.36:53611 - - [01/Mar/2020:14:24:23 -0500] "GET /?message=ERROR%3A%20Process%20scoreboard%3A%20no%20such%20file HTTP/1.1" 200 2645
2020-03-01 09:24:32,650 ERRO 192.168.1.36:53611 - - [01/Mar/2020:14:24:32 -0500] "GET /?message=ERROR%3A%20Process%20scoreboard%3A%20no%20such%20file HTTP/1.1" 200 2645
2020-03-01 09:30:52,012 CRIT Supervisor is running as root.  Privileges were not dropped because no user is specified in the config file.  If you intend to run as root, you can set user=root in the config file to avoid this message.
2020-03-01 09:30:52,019 INFO Included extra file "/etc/supervisor/conf.d/scoreboard.conf" during parsing
2020-03-01 09:30:52,201 INFO RPC interface 'supervisor' initialized
2020-03-01 09:30:52,208 CRIT Server 'inet_http_server' running without any HTTP authentication checking
2020-03-01 09:30:52,214 INFO RPC interface 'supervisor' initialized
2020-03-01 09:30:52,218 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2020-03-01 09:30:52,223 INFO supervisord started with pid 415
2020-03-01 09:30:52,233 INFO spawnerr: can't find command '[sudo'
2020-03-01 09:31:04,451 INFO spawnerr: can't find command '[sudo'
2020-03-01 09:31:06,633 INFO spawnerr: can't find command '[sudo'
2020-03-01 09:31:09,654 INFO spawnerr: can't find command '[sudo'
2020-03-01 09:31:09,670 INFO gave up: scoreboard entered FATAL state, too many start retries too quickly
2020-03-01 09:34:51,253 CRIT Supervisor is running as root.  Privileges were not dropped because no user is specified in the config file.  If you intend to run as root, you can set user=root in the config file to avoid this message.
2020-03-01 09:34:51,268 INFO Included extra file "/etc/supervisor/conf.d/scoreboard.conf" during parsing
2020-03-01 09:34:52,158 INFO RPC interface 'supervisor' initialized
2020-03-01 09:34:52,171 CRIT Server 'inet_http_server' running without any HTTP authentication checking
2020-03-01 09:34:52,193 INFO RPC interface 'supervisor' initialized
2020-03-01 09:34:52,205 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2020-03-01 09:34:52,218 INFO supervisord started with pid 419
2020-03-01 09:34:52,281 INFO spawned: 'scoreboard' with pid 430
2020-03-01 09:34:53,502 INFO success: scoreboard entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-03-01 09:35:12,895 INFO exited: scoreboard (exit status 2; expected)
2020-03-01 09:35:13,496 INFO spawned: 'scoreboard' with pid 439
2020-03-01 09:35:14,540 INFO success: scoreboard entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-03-01 09:35:31,039 INFO exited: scoreboard (exit status 2; expected)
2020-03-01 09:35:32,113 INFO spawned: 'scoreboard' with pid 448
2020-03-01 09:35:33,173 INFO success: scoreboard entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-03-01 09:35:49,473 INFO exited: scoreboard (exit status 2; expected)
2020-03-01 09:35:50,555 INFO spawned: 'scoreboard' with pid 457
2020-03-01 09:35:51,613 INFO success: scoreboard entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-03-01 09:36:07,819 INFO exited: scoreboard (exit status 2; expected)
2020-03-01 09:36:08,894 INFO spawned: 'scoreboard' with pid 466
2020-03-01 09:36:09,952 INFO success: scoreboard entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-03-01 09:36:26,580 INFO exited: scoreboard (exit status 2; expected)
2020-03-01 09:36:27,655 INFO spawned: 'scoreboard' with pid 475
2020-03-01 09:36:28,716 INFO success: scoreboard entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-03-01 09:36:45,189 INFO exited: scoreboard (exit status 2; expected)
2020-03-01 09:36:46,264 INFO spawned: 'scoreboard' with pid 484

falkyre, 2020-03-01:
In your scoreboard.conf command get rid of the sudo.

falkyre, 2020-03-01:
So make it command=python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=65 --led-slowdown-gpio=2

jeepdrvr71, 2020-03-01:
[program:scoreboard]
command=python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=65 --led-slowdowsn-gpio=2
directory=/home/pi/nhl-led-scoreboard
autostart=true
autorestart=true

This is what I am showing. I rebooted and still have a blank screen.

jeepdrvr71, 2020-03-01:
I ended up changing the scoreboard.conf file. I set autostart=false and autorestart=false and re-installed the old way that I had to start it up. Everything came up fine that way. 

I am going to grab another sd card and install everything fresh and use the supervisor. I think there was something that got written to a file that was messing everything up. 

Thanks again for the help falkyre. You guys are awesome on this board. I hope to contribute some things along the way. I am new to all this and trying to get up to speed.

riffnshred, 2020-03-02:
*BLUE YOU'RE MY BOY*
Thank you so much falkyre. jeepdrvr71, let us know if that fixed it and close the issue if it did

closed 2020-03-02 by jeepdrvr71