riffnshred/nhl-led-scoreboard #62: src/boards/team_summary.py line 98: "gamesPlayed" is not an attribute of stats?

ka9cql, 2020-03-02:
nhl-led-scoreboard/src/boards/team_summary.py", line 98, in draw_team_summary
    draw.text((0, 7), "GP: {} P: {}".format(stats.gamesPlayed, stats.pts), fill=(255, 255, 255),
AttributeError: 'bool' object has no attribute 'gamesPlayed'

I can not find "gamesPlayed" anywhere in the codebase other than this line.  Either my codebase is out of date (I was on beta, but force-pulled the main) or this is just an error.  Perhaps it's a spelling error??
[gamesPlayed_error.txt](https://github.com/riffnshred/nhl-led-scoreboard/files/4277953/gamesPlayed_error.txt)

ka9cql, 2020-03-02:
I think it's "bool" because - in the absence of the NHL API returning any stats for your team, line 47 of src/nhl_api/info.py sets "stats = False".

If so, then line 98 of src/boards/team_summary.py probably needs a check whether "stats" is a boolean (False, as in the case of no stats for that team) or a "stats" dictionary or list, as picked up from the NHL API on line 44 of src/nhl_api/info.py

ka9cql, 2020-03-02:
Found this, might be a good fix for line 98 on src/boards/team_summary.py -
if isinstance(stats, bool) == False:
    draw.text((0, 7), "GP: {} P: {}".format(stats.gamesPlayed, stats.pts), fill=(255, 255, 255),........(etc)

This fix (test-stats-for-NOT-boolean) should protect lines 96-101, or something, to prevent other uses of the "stats" variable when it's not filled in (aka boolean set to False).  I just don't know the codebase well enough to formally submit a "complete" fix...

riffnshred, 2020-03-22:
This was cause by an error in the way I printed the error message. Got fix in V 1.1.2

closed 2020-03-22 by riffnshred