riffnshred/nhl-led-scoreboard #69: Game board didn’t start until 17:33 left in the first

cipriani14, 2020-03-05:
Not sure why there was a delay last night, but the game board didn’t start in sync with the game. First time it’s happened. Here’s my config file. 

 
[image]
[image]

cipriani14, 2020-03-09:
So I switched my top team to the Blues earlier tonight and everything worked as it should. I switched back to the Avs before the game started, and the scoreboard didn’t switch to game mode until 17:35. Here’s the log:

`INFO (20:08:18): Using software to determine dimming
INFO (20:08:18): Rendering...
INFO (20:08:18): Checking for dimmer
refresing data
INFO (20:08:18): Using software to determine dimming
INFO (20:08:18): It is night time
INFO (20:08:19): Game Day Wooooo
INFO (20:08:19): Showing Game
INFO (20:08:19): Checking for new day
INFO (20:08:19): It is not a new day
INFO (20:08:19): Game is Scheduled
INFO (20:08:19): Showing Main Event
INFO (20:11:58): Checking for new day
INFO (20:11:58): It is not a new day
refreshing
INFO (20:11:58): Game is Scheduled
INFO (20:11:58): Showing Main Event
INFO (20:13:18): Checking for dimmer
INFO (20:13:18): Using software to determine dimming
INFO (20:13:18): It is night time
INFO (20:15:37): Checking for new day
INFO (20:15:37): It is not a new day
refreshing
INFO (20:15:38): Game is Scheduled
INFO (20:15:38): Showing Main Event
INFO (20:18:18): Checking for dimmer
INFO (20:18:18): Using software to determine dimming
INFO (20:18:18): It is night time
INFO (20:19:17): Checking for new day
INFO (20:19:17): It is not a new day
refreshing
INFO (20:19:17): Game is Live
DEBUG (20:19:17): Check new goal
INFO (20:19:17): Showing Main Event
INFO (20:19:27): Checking for new day
INFO (20:19:27): It is not a new day
refreshing
INFO (20:19:27): Game is Live
DEBUG (20:19:27): Check new goal
INFO (20:19:27): Showing Main Event
INFO (20:19:37): Checking for new day
INFO (20:19:37): It is not a new day
refreshing
INFO (20:19:38): Game is Live `

riffnshred, 2020-03-09:
This is likely caused by a timing factor. In your config, I see you are showing all the boards and also both conference wild_card standings. It seems to take about 3 min to go through them. What I think is going on is, the scoreboard refreshed the data a few seconds before the game started, so didn't catch it, and then it started to show the list of boards. Once done, it refreshes the data and then catches the game which already started about 2 min 30 sec ago.

As discussed on discord last night, I'll try to run the data module in a thread and as soon as it detect that the game started,  stop the board rotation and go to live mode.