riffnshred/nhl-led-scoreboard #71: ModuleNotFoundError: No module named 'ephem'

ThePapatech, 2020-03-08:
I have run through the instructions multiple time and have reinstalled pip but still I get this error when running the scoreboard:

pi@scoreboard:~/nhl-led-scoreboard $ sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2
Traceback (most recent call last):
  File "src/main.py", line 9, in <module>
    from dimmer import Dimmer
  File "/home/pi/nhl-led-scoreboard/src/dimmer.py", line 1, in <module>
    import ephem
ModuleNotFoundError: No module named 'ephem'

I have been chasing it all day and have googled myself into many blackholes! Any help is appreciated.

riffnshred, 2020-03-08:
You installed PIP or PIP3?

ThePapatech, 2020-03-08:
I have tried both...here is what it came back with last time...

pi@scoreboard:~ $ sudo apt install git python3-pip
Reading package lists... Done
Building dependency tree
Reading state information... Done
git is already the newest version (1:2.20.1-2+deb10u1).
python3-pip is already the newest version (18.1-5+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 15 not upgraded.

joshkay, 2020-03-08:
Try running the install script again.

M4lwar, 2020-03-08:
I had same issue, had to manually install each package calling pip3.

ThePapatech, 2020-03-08:
I will try the script again.

ThePapatech, 2020-03-08:
I re-ran the script again and now it shows the scoreboard. Not sure what made it work but thanks for the push of ideas to get me back on track!

closed 2020-03-08 by ThePapatech

riffnshred, 2020-03-08:
Glad it finally worked. Enjoy!!