riffnshred/nhl-led-scoreboard #74: Scoreboard locked up

jeepdrvr71, 2020-03-13:
Now that the season has been suspended my scoreboard froze yesterday at 1:43pm and stayed on that. I tried cycling power but now it's blank.

I assume it's because no data is coming from the NHL API. Maybe a change could be made to default to a clock mode if no data is available.

riffnshred, 2020-03-13:
Hey man, Ya I didn program the scoreboard to handle the posponed games yet, and I was not expecting that to happen anytime soon. I'm working on a fix and alternatives ATM. API is fine at least.

jeepdrvr71, 2020-03-13:
I wasn't implying it was a bug. No one could foresee this. I was just thinking it may be a good thing to program in for even the off season. So that way if there are no games it can show the clock at least.

riffnshred, 2020-03-13:
I'm at work so I can't test that but try this out. Put the live mode to
False in the config and only put the clock in the Off_day state also in the
config file. I think that will run the clock 24/7
FYI I'm working on a COVID 19 counter for LED matrix. I'll release that soon

jeepdrvr71, 2020-03-13:
That is great idea.

jeepdrvr71, 2020-03-18:
Have you got any further with your Covid 19 code?

riffnshred, 2020-03-22:
Just release, fresh out of the oven. update your board and also, you can use the new app to configure your board. if you need help, come see us in the discord channel.

cheers

closed 2020-03-22 by riffnshred