riffnshred/nhl-led-scoreboard #77: Scripts to Swap between MLB and NHL boards

necromancyr, 2020-03-17:
So, this is a very rough, crappy python and bash script to toggle between the MLB and NHL scoreboards (and shutdown if a button is held).  This is something I just cobbled together and I'm certain there's a better way to do it.  

Key things I did:
- Uses the existence/creation of a directory in a bash script to toggle between MLB and NHL.  This only works for 2 options and is definitely 'dirty'.   There's no reason the individual commands can't be pulled into the python script itself and called using a more elegant setup.
- Everything is obviously hardcoded.  

My next step from here is putting everything into a python script only that could switch through more options, isn't so kludgey, etc.

boardtoggle.sh
`#!/bin/bash

if mkdir .togglestate
then
	echo "-- .togglestate doesn't exist.  Making."
	echo "- Kill MLB, Start NHL"
	echo "- Killing All"
	sudo supervisorctl stop MLB
	echo "- Starting NHL"
	sudo supervisorctl start NHL
else
	echo "-- .togglestate can't be made."
	echo "- Kill NHL, Start MLB"
	echo "- Removing .togglestate"
	rmdir .togglestate
        echo "- Killing All"
	sudo supervisorctl stop NHL
        echo "Starting MLB"
	sudo supervisorctl start MLB
fi
`

Button Press Detection
`from gpiozero import Button
from signal import pause
import time
from subprocess import check_call
from subprocess import call
import os

boardswapcmd = "/home/pi/boardtoggle.sh"

Button.was_held = False

def held(btn):
    btn.was_held = True
    print("Button hold detected.  Shutting down.")
    time.sleep(1)
    print("-----SHUTDOWN COMMAND HERE-----")
    os.system("sudo shutdown -h now")

def released(btn):
    if not btn.was_held:
        pressed()
    time.sleep(1)
    check_call(boardswapcmd)
    btn.was_held = False

def pressed():
    print("Button press detected.  Swapping sports.")

btn = Button(3, hold_time=5)

btn.when_held = held
btn.when_released = released

pause()
`

ajbowler, 2020-03-17:
Hey @necromancyr I'm the MLB board creator. This looks really interesting! You might consider making this a separate project since it's basically its own tool that facilitates swapping between the two boards. I'd be happy to link it in the MLB board README and wiki once it's up and running

necromancyr, 2020-03-18:
Good idea.  :)

https://github.com/necromancyr/swap-led-scoreboard/blob/master/README.md

Modified into one python script that, theoretically, you can setup as a always running under supervisor that will toggle on a short press (w/1 second debounce in there) and shutdown on a custom configurable hold.  

Also, you need to set the names of your supervisor configs for each board - theoretically an infinite number would be possible.

jeepdrvr71, 2020-03-25:
Does anyone have both the MLB and NHL project running on their units and is switching using the Supervisor?

Also has anyone thought of doing something similar for the English Premier League? I would like to try doing it, but my programming skills are limited. And to tell you the truth I don't know where to start.

machinasdei, 2021-01-03:
Hello all,
here's a solution to this request that works for me on a pi zero w - a simple bash script piticker.sh - i'm running 4 different boards and it cycles through each one..... just edit the sleep time you want.  check it out at:

https://pastebin.com/sVx8chS1

riffnshred, 2021-01-04:
Sweet. thanks for that !!

closed 2021-03-24 by riffnshred

pjockey, 2021-03-26:
Looking for some information on how to add a second instance into the /etc/supervisor/conf.d/scoreboard.conf

falkyre, 2021-03-26:
@pjockey just add another file in the conf.d named something different and make sure you change the program name in the new conf file.  Restart supervisor after you add the new file so it reloads it. 

So your scoreboard.conf has this at the top of the file. 

[program:scoreboard]

If you make a new conf file called hockey.conf make sure your top line is:

[program:hockey]

pjockey, 2021-03-26:
This what I have:

```
[program:Scoreboard]
command=sudo python3 src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2
directory=/home/pi/nhl-led-scoreboard
command=sudo python main.py --gpio-maping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2 --led-rows=32 --led-cols=64
directory=/home/pi/mlb-led-scoreboard
autostart=true
autorestart=true
```

falkyre, 2021-03-27:
You need to have two conf files with a single command in each file.   You have two command stanzas in your file and you only need one.  Also, your command doesn't need Sudo as supervisor already runs everything as root user.

pjockey, 2021-04-02:
Could someone post the contents of the /etc/supervisor/supervisord.conf file? Thanks. I've somehow messed the innards of the file up. 

I can't get the web interface to work after uninstalling and re-installing supervisor.

riffnshred, 2021-04-02:
check it out in the wiki section. it's under the step by step guide

pjockey, 2021-04-03:
Those 2 lines is all that is in the file? Been following the wiki, step by step, and I can't get the browser to recognize anything.