riffnshred/nhl-led-scoreboard #80: Config file error

desigi, 2020-03-22:
When running this command:
`sudo python3 /home/pi/nhl-led-scoreboard/src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=30 --led-slowdown-gpio=2`

I get the following error:
```
INFO (01:47:44): NHL-LED-SCOREBOARD - v1.1.2 (64x32)
Config file config/config.json not found!
Config file config/fonts/fonts.json not found!
Config file config/fonts/fonts_64x32.json not found!
Traceback (most recent call last):
  File "/home/pi/nhl-led-scoreboard/src/main.py", line 60, in <module>
    run()
  File "/home/pi/nhl-led-scoreboard/src/main.py", line 34, in run
    config = ScoreboardConfig("config", commandArgs, (matrix.width, matrix.height))
  File "/home/pi/nhl-led-scoreboard/src/data/scoreboard_config.py", line 87, in __init__
    self.config = Config(size)
  File "/home/pi/nhl-led-scoreboard/src/config/main.py", line 9, in __init__
    self.fonts = FontsConfig(size)
  File "/home/pi/nhl-led-scoreboard/src/config/files/fonts.py", line 12, in __init__
    self.load_fonts()
  File "/home/pi/nhl-led-scoreboard/src/config/files/fonts.py", line 17, in load_fonts
    for element, value in self.config.data:
AttributeError: 'ConfigFile' object has no attribute 'data'
```
I ran the `nhl_setup` script and it saved the config file correctly. Any idea what's causing this issue. The command running in Supervisor is also returning this error.

riffnshred, 2020-03-22:
Not sure why, but if you run the command directly from the nhl-led-scoreboard folder, it works as usual.

riffnshred, 2020-03-22:
may I ask why you run from the root of the Pi instead of the root of the software ?

long-horse, 2020-03-23:
I'm also getting this error even when running `sudo python3 src/main.py --led-gpio-mapping=adafruit-hat --led-brightness=60 --led-slowdown-gpio=2` from `~/nhl-led-scoreboard`

riffnshred, 2020-03-23:
I'll have a look today. The nhl_setup app uses the sample to create your config.json, not sure if you fooled around with the sample and that might be the cause of the app not running properly.  try to re-run the install.sh and after run `./nhl_setup`. That worked for everybody so far.

desigi, 2020-03-23:
I think I found the source of the error. It seems to come from the config file locations set in the `fonts.py`, `layout.py`, and `settings.py` files.  The json file locations are set to `config/config.json` or `config/layout/layout.json` etc.
 
It seems when you run the application anywhere out side the project folder it can't find the files. (might be looking for them in `nhl-led-scoreboard/src/config/` instead of `nhl-led-scoreboard/config`.

I set the absolute path in each of they python files and it worked for me. I used `/home/pi/nhl-led-scoreboard/config/config.json` instead of `config/config.json`.

riffnshred, 2020-03-23:
Ya we found the issue last night and it is coming from the scoreboard_config.py that looks for those folders specifically. Is there a reason you run from the root of the Pi instead of the root of the software ?. The issue with your solution is that everytime you will update, you will have to hard code the location again. 

We might fix it so you can run the software from anywhere sooner or later, but for now, I recommend running from the software directory as described in the documentation.

desigi, 2020-03-23:
I was having an issue getting Supervisor to run from this project folder. I figured it out though. I just needed to restart the supervisor service and everything is working now.
Apologies for the inconvenience.

closed 2020-03-23 by desigi

riffnshred, 2020-03-23:
All good bud. No inconvenience. The new way to configure sparked a lot of confusion and issues last night on discord. I should have reported our find in here also. Thanks for reporting man !!!

riffnshred, 2020-03-23:
Also !!! Someone pointed out, the setup app shouldn't be run with sudo.

desigi, 2020-03-23:
Without sudo I would get an error at the end to write the json file.

riffnshred, 2020-03-23:
> `Traceback (most recent call last): File "nhl_setup.py", line 638, in <module> File "nhl_setup.py", line 193, in main File "nhl_setup.py", line 59, in load_config File "json/__init__.py", line 296, in load File "json/__init__.py", line 348, in loads File "json/decoder.py", line 337, in decode File "json/decoder.py", line 353, in raw_decode json.decoder.JSONDecodeError: Expecting ',' delimiter: line 69 column 20 (char 1075) [17959] Failed to execute script nhl_setup`

That error happened because the "Pin" option in the Pushbutton section in the config.json.sample was changed and the ',' was deleted at the same time. That's what this error points out (I didn't get that my self, a kind person pointed it out to me). That explains the error you got while running the nhl_setup app.

long-horse, 2020-03-24:
I just reinstalled buster because I was on a noobs install. 

I did `./nhl_setup` and at the end it said `Error making backup of config/config.json. Error: [Errno 2] No such file or directory: 'config/config.json'`

I also do not see the net,stick and horn on the matrix when first running it. Sorry to hijack, not sure if I should open a new case.

riffnshred, 2020-03-24:
This is different bud, and thanks for opening issue #82