riffnshred/nhl-led-scoreboard #81: Enhancement Team colours for the clock

dmryglod, 2020-03-22:
Modifying the layout so team colours could be assigned to the clock fonts. You can get the team colours by changing things in the layout.json, but modifying the default colour cascades it throughout the entire display when the default colour is used. A tag for the primary and secondary colour in the clock layout would address this.

riffnshred, 2020-03-24:
Love the idea to use team's colors for the clock and this could be an option to add. For now tho there is an easier way then hard coding it. Under config/colors/layout.json there is color settings for the clock that has `"time": [255, 50, 50]`. underneath it add a color setting for `"date"` and for `"meridiem"`. 

your config should look like this. 
```
{
  "_default": [255, 255, 255],
  "clock": {
    "time": [255, 50, 50],
    "date": [255, 50, 50],
    "meridiem": [255, 220, 50]
  },
  "scoreboard": {
    "period_final": [255, 0, 0],
    "SOG_label": [255, 111, 0],
    "irregular_status":[255, 0, 0]
  },
  "covid_19": {
    "name": [0, 0, 0],
    "location": [255, 255, 255],
    "board_title":[0, 0, 0]
  }
}
```

Just change the RGB color code to your team's colors. ;)

riffnshred, 2021-01-04:
This was added by @falkyre in v1.5.0.

closed 2021-01-04 by riffnshred