riffnshred/nhl-led-scoreboard #82: config file error

long-horse, 2020-03-24:
I took the liberty to document everything I have done up until my error so whoever is kind enough to help me can see exactly what I have done.

I installed Raspbian Buster Lite with Etcher
In the config, I set my password, hostname, wifi (ssid&pass), timezone, wifi country, ssh, disabled one-wire (I saw this was recommended but between MLB, NHL and Matrix gits, I can't remember where), & expanded filesystem

Now, via ssh (after clearing the old SSH handshake key), I ran

`sudo apt-get update
sudo apt install git python3-pip`

then I rebooted and ran 

`git clone --recursive https://github.com/riffnshred/nhl-led-scoreboard
cd nhl-led-scoreboard/
sudo chmod +x scripts/install.sh
sudo ./scripts/install.sh`

At the very bottom of the long wall of install text:

`If you didn't see any errors above, everything should be installed!
Installation complete! Play around with the examples in nhl-led-scoreboard/submodules/matrix/bindings/python/samples to make sure your matrix is working.`

At some point in the long wall of install text, I saw an error. The first 2 lines in green followed by the 3rd line in red (presumably denoting an error):

`Installing collected packages: rgbmatrix
  Found existing installation: rgbmatrix 0.0.1
Cannot uninstall 'rgbmatrix'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.`

I am not sure how rgbmatrix is installed on a fresh install of raspbian buster lite...Or does the installer try to install it twice? It can't possibly have remained after reflashing the OS, causing confusion.

So far this is all straight from the [main page](https://github.com/riffnshred/nhl-led-scoreboard). Someone, please correct me if I made a mistake so far. 

The error with rgb matrix did not seem related to the config file error so I ignored and continued.

Next it says to head to the config section, which says to run `./nhl_setup` from `~/nhl-led-scoreboard` on the new V1.1.2 which I am sure I am running as this is installing at March 24, 2020, 12:24 AM EST. 

To repeat, I am looking at `pi@raspberryty:~/nhl-led-scoreboard $ ` and I ran `./nhl_setup`

I went through the config and at the end I am greeted with 
`Error making backup of config/config.json. Error: [Errno 2] No such file or directory: 'config/config.json'`

I can copy and paste all my config options if that will help.

joshkay, 2020-03-24:
I believe that error is expected during the first run. It's just saying it was unable to backup the config because there wasn't an existing file. You should be able to proceed with the instructions on how to run the scoreboard.

long-horse, 2020-03-24:
Ok. I am baffled. The scoreboard actually works. I've reinstalled fresh twice and did the same steps and it didn't work. And now it does. I feel pretty dumb right about now. Even supervisor is restarting it when it ends...

long-horse, 2020-03-24:
I am having trouble with my rotation. I am seeing today's games and COVID-19. I am not seeing standings or clock.

riffnshred, 2020-03-24:
no worries man. Can you copy and paste the content of your config.json ?

long-horse, 2020-03-24:
{
    "debug": true,
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "8:00",
        "live_game_refresh_rate": 10,
        "teams": [
            "Rangers",
            "Islanders"
        ],
        "sog_display_frequency": 4
    },
    "goal_animations": {
        "pref_team_only": true
    },
    "states": {
        "off_day": [
            "scoreticker",
            "covid_19",
            "team_summary",
            "standings",
            "clock"
        ],
        "scheduled": [
            "standings",
            "team_summary"
        ],
        "intermission": [
            "scoreticker"
        ],
        "post_game": [
            "scoreticker",
            "standings"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "division",
            "divisions": "metropolitan",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false
        }
    },
    "sbio": {
        "dimmer": {
            "enabled": false,
            "source": "software",
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": false,
            "bonnet": true,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "clock",
            "state_triggered1_process": ""
        }
    }
}

long-horse, 2020-03-24:
It seems to only rotate between 2 of the off day states. Initially it was score ticker and covid19 but if I rearranged them I did see what the others look like...2 at a time. Looks really good! I just hope I can rotate between 3 or 4.

riffnshred, 2020-03-24:
I just tested your config and it crashed after the second board. The reason why you only see 2 boards is because the scoreboard crash after the second and supervisor reboot the software after the crash.

Thanks for reporting, I'll work on it today and see if it's an actual bug or if it's something in the config you have.

falkyre, 2020-03-24:
> I believe that error is expected during the first run. It's just saying it was unable to backup the config because there wasn't an existing file. You should be able to proceed with the instructions on how to run the scoreboard.

Josh is right.  On first run, the nhl_setup app will take the config.json.sample and use it as the defaults to ask you the questions.  When it's done and you are asked if you want to save, the app will try and make a backup.  If the file doesn't exist, the copy fails and you get that error message:

Error making backup of config/config.json. Error: [Errno 2] No such file or directory: 'config/config.json

This is ok and will not cause any issues and this is running by design.  Your other issue about permissions (I believethe Error 13) is likely because the config.json has ben editing/created using a sudo command.  The config.json doesn't need to be edited using sudo but if you have, you will need to run the nhl_setup using the sudo command so it can read your config.

Finally, you also mentioned about not seeing the net, stick and horn.  This, again, is by design and will not cause any issues.  See the readme for the app: [nhl_setup Readme]( https://github.com/riffnshred/nhl-led-scoreboard/blob/master/src/nhl_setup/README.md)

long-horse, 2020-03-24:
@riffnshred ive been away from this type of thing for too long to remember basic diagnosis. Yes, it was (now obviously) crashing on the off day state...Today the 24th both of my selected teams are playing and it is working properly. But tomorrow it will break again :(

riffnshred, 2020-03-24:
@long-horse  Turns out it was a bug in the board rotation, Thanks for finding this !! Fixed it in V1.1.3 !!

closed 2020-03-24 by riffnshred

long-horse, 2020-03-26:
Hey, I ran the 2 commands from main NHL dir to update but it didn't look like it updated anything. Is there a way to check my version? 

`sudo chmod +x scripts/install.sh`
`sudo ./scripts/install.sh`

riffnshred, 2020-03-26:
I don't have a script to confirm which version of the software is install at the moment but run this git command from the nhl scoreboard root directory.

`git rev-parse HEAD`

This will give the hash version of the current Master. At the moment it starts with 47f549f6..... if that's what you get then you have the right version.

desigi, 2020-03-26:
The version gets out put to the terminal when the `main.py` file is run. It is also in the first few lines of the `main.py` file if you view the contents of the file.