riffnshred/nhl-led-scoreboard #85: COVID_19 board time 12/24 hour

ruchoff, 2020-03-25:
**Description**
Have the time displayed on the covid_19 board be displayed in the time format set in the time_format config variable.

jeepdrvr71, 2020-03-26:
I have a red line that is now popping up on the bottom of the Covid-19 screeens. Is this some kind of indicator?

jeepdrvr71, 2020-03-26:
FYI-I restarted the program and the red line went away.

riffnshred, 2020-03-26:
Its the network issue indicator. It was on for about 40min because the Covid 19 API was down.

ntjoumas, 2020-03-31:
I have added in support to utilize format for covid_19 board from config.json. Will issue PR for changes in dev branch shortly.

riffnshred, 2021-03-24:
the covid 19 board is no more since V1.6.0 as this was a temporary feature after the NHL 2020 season got canceled.

closed 2021-03-24 by riffnshred