riffnshred/nhl-led-scoreboard #95: Power Button

stufo19, 2020-04-09:
I am having trouble getting the power button feature to work. Do I need to install some sort of script in order to get this feature to work?

riffnshred, 2020-04-09:
I'll need some info to help you.

1. have you fallowed the instructions from this Repo or from Howchoo ?
2. Do you use the HAT or the Bonnet
3. Which pins do you have you button connected to ?

stufo19, 2020-04-09:
1. I have followed the instructions from this repo.
2. I am using the bonnet.
3. I have the button connected to the ground pin and pin 3, the scl pin.

riffnshred, 2020-04-09:
can you send the pushbutton section of you config.json ?

stufo19, 2020-04-09:
"pushbutton": {
            "enabled": true,
            "bonnet": true,
            "pin": 3,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 5,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "covid_19",
            "state_triggered1_process": ""

riffnshred, 2020-04-09:
Config looks good. so when you say trouble, do you mean not working at all? Does if shutdown after 5 seconds ?

stufo19, 2020-04-09:
Yes, I tested my button with a multimeter and it appears to work so I'm not sure why it doesn't work.

riffnshred, 2020-04-09:
To put it on record and to answer your original question, you don't need any script to use the power on and off function. The SBIO module take care of it. 

In your situation I'd re run the install.sh and the ./nhl_setup. If that does not fix in. send a picture of your bonnet. I'll get some more info from the contributor who wrote the SBIO module.

riffnshred, 2020-04-09:
Here's something to test out your button outside the software. 
load this script onto your Pi into the root of the nhl-led-scoreboard. 

https://gofile.io/?c=wcwcfe

Than run it `sudo python3 test_button_gpio3.py`. this will read the GPIO 3 pin every 0.1 seconds. if the button is pressed it will print "Button is pressed" on your terminal, otherwise it will print "Button is not pressed".

stufo19, 2020-04-10:
It turns out that re running the scripts and redoing some soldering fixed the issue, thank you.

closed 2020-04-10 by stufo19