riffnshred/nhl-led-scoreboard #97: Option to disable animation on clock

harmgsn, 2020-05-28:
In config.json it would be nice to have an option to disable the blinking cursor on the clock board.  Right now I've been modifying line 27 of src/boards/clock.py from:

            self.time = datetime.datetime.now().strftime(self.time_format.replace(":", " "))

to

            self.time = datetime.datetime.now().strftime(self.time_format.replace(":", ":"))

This effectively disabled the clock, but with each update I have to remember how I did this and redo it.   I didn't see an option anywhere else to disable this, so this seemed like the easy mode to take care of it.

riffnshred, 2020-05-29:
I'm not going to bother for now since the way we config stuff is not settle yet  and there is a lot of work behind coding config options for such a small thing. You can note where the file and the line is for future reference.

Ill keep this in mind when we get around making an easier way to config options.

falkyre, 2020-09-30:
This is part of PR #144

closed 2021-01-03 by riffnshred