riffnshred/nhl-led-scoreboard #98: Customize clock color

CocoCat2, 2020-06-05:
How would one go about changing the colors on the clock board? I’d like to be able to customize the colors of the various fields.  (Month, day, year, hour, min, etc)

riffnshred, 2020-06-05:
If you go to, nhl-led-scoreboard/config/colors/layout.json, the colors are in this array of RGB colors:

"clock": {
    "time": [255, 50, 50],
    "date": [255, 255, 255],
    "meridiem": [255, 255, 255]
},

cheers

CocoCat2, 2020-06-05:
Thanks, but I want to customize the colors of the individual segments.  As in, the month being one color, the day being another, the year yet another.  Could I add a “month”:[255,255,0], hour[100,255,50] etc.. in there to do what I want? Or would it require a different way to pull the data in clock.py? — I have just enough programming knowledge to fubar things, but not enough to do what I want to accomplish lol

riffnshred, 2020-06-05:
Oooooh !!! Currently the code for the clock needs adjustment for this but nothing too fancy. I'll add that for an up coming update.

falkyre, 2020-09-30:
Most of this is in PR #144.  The clock can now use your first preferred team colors or you can set the RGB of the clock and date numbers

closed 2021-01-03 by riffnshred