riffnshred/nhl-led-scoreboard #102: Bug in the nhl_setup script update that comes with the weather feature.

riffnshred, 2020-06-16:
**Describe the bug**
@falkyre hey man, just a heads up, there is a bug in your new setup script. When I select the quick "select one team" setup it does not put the name of the team as a table with `[]` . That's what the software expect when it checks for the preferred team. That also bring up an error in when I rerun the setup again but this time I do the complete one and right when it gets to selecting the teams it crash.

**Logs**
1.3.0 - Wx version
🥅🏒🚨 Do you see a net,stick and horn?  Yes                                                                                                                         
🥅 Do you want a simple default setup with one team selection (Y) or full setup (N)?  No                                                                             
🥅 Do you want debug on? (outputs log messages to console)  Yes                                                                                                      
🥅 Do you want live_mode on? (Shows live game data of favorite team)  No                                                                                             
🥅 Select the time format  24h                                                                                                                                       
🥅 When do you want end of day to be? 8:00                                                                                                                           
🥅 Your location to override latitude and longitude lookup via IP (City, State or City, Province) Winnipeg, MB                                                       
🥅 Live game refresh rate? (rate at which a live game will call the NHL API to catch the new data, 10 seconds is the lowest) 10                                      
Traceback (most recent call last):
  File "nhl_setup.py", line 1140, in <module>
  File "nhl_setup.py", line 523, in main
  File "nhl_setup.py", line 202, in get_team
TypeError: can only concatenate str (not "list") to str
[19722] Failed to execute script nhl_setup

falkyre, 2020-06-16:
Hmmm. Ok. I'll take a look.  Just finishing up something else right now.

falkyre, 2020-06-16:
What happens when you put the team on the command line? Same thing?

riffnshred, 2020-06-16:
curiously its fine when I do that !!!

falkyre, 2020-06-17:
OK, have a fix and some bugs I squashed for the weather stuff.  Do you want another PR against beta?

falkyre, 2020-06-17:
See PR #104 for fix #

closed 2020-06-18 by riffnshred