riffnshred/nhl-led-scoreboard #109: Display Dead After Uploading Version with Weather

jeepdrvr71, 2020-07-01:
I installed everything and didn't get any errors. I updated the config.json and restarted using the supervisor and got nothing.
Here is my config.json. Any help would be appreciated.

 "debug": false,
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "3:00",
        "live_game_refresh_rate": 15,
        "teams": [
            "Canadiens"
        ],
        "sog_display_frequency": 5
    },
    "goal_animations": {
        "pref_team_only": true
    },
    "states": {
        "off_day": [
            "clock",
            "scoreticker",
            "standings",
            "team_summary",
            "weather"
        ],
        "scheduled": [
            "clock",
            "weather"
        ],
        "intermission": [
            "clock",
            "scoreticker",
            "weather"
        ],
        "post_game": [
            "clock",
            "scoreticker",
            "standings",
            "team_summary",
            "weather"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "wild_card",
            "divisions": "atlantic",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false
 }
        },
        "weather": {
                "enabled":true,
                "units":"imperial",
                "duration":"30",
                "data_feed":,"OWM",
                "alert_feed":"NWS",
                "owm_apikey":""a3f5851658a8714b7be4f880db684a92",
                "update_freq":"5",
                "show_alerts":true,
                "alert_title":true,
                "scroll_alert":true,
                "alert_duration":"5",
                "show_on_clock":false
        }

 },
    "sbio": {
        "dimmer": {
            "enabled": true,
            "source": "software",
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 40,
            "sunrise_brightness": 80
        },
        "pushbutton": {
            "enabled": true,
            "bonnet": true,
            "pin": 25,

riffnshred, 2020-07-01:
any error message from the raspberry pi ? in supervisor if you click on the name of the process you can see the logs. copy and paste it here. 

There are a couple of issues going on with the latest update and we have an update coming tomorrow that should fix them.

jeepdrvr71, 2020-07-01:
No errors on the pi and when I click on the process in supervisor it goes to a new tab and only has a refresh button showing. If there is an update coming tomorrow I will wait till then and try updating again.

cipriani14, 2020-07-01:
That’s the same thing that happens to me.

riffnshred, 2020-07-02:
we are working on a fix, I'll keep you posted boys !

dawelge01, 2020-07-02:
I wondering if it needs coordinates?

riffnshred, 2020-07-03:
Just pushed an update that should fix this issue. can you confirm @jeepdrvr71 ?

jeepdrvr71, 2020-07-03:
I have upgraded using the new update. I still don't have a display but at least there is something in the logs. Here they are.

INFO (07:59:33): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
ERROR (07:59:33): Unable to load json: Expecting value: line 61 column 15 (char 1341)
INFO (07:59:47): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
ERROR (07:59:47): Unable to load json: Expecting value: line 61 column 15 (char 1341)
INFO (08:00:00): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
ERROR (08:00:00): Unable to load json: Expecting value: line 61 column 15 (char 1341)
INFO (08:00:14): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
ERROR (08:00:14): Unable to load json: Expecting value: line 61 column 15 (char 1341)
INFO (08:00:27): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
ERROR (08:00:27): Unable to load json: Expecting value: line 61 column 15 (char 1341)
INFO (08:00:40): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
ERROR (08:00:40): Unable to load json: Expecting value: line 61 column 15 (char 1341)
INFO (08:00:53): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
ERROR (08:00:54): Unable to load json: Expecting value: line 61 column 15 (char 1341)

My config.json is in a previous post.

riffnshred, 2020-07-03:
looks like your config is missing a chunk of stuff. you'll have to re-configure. delete your config.json and run `./nhl_setup` do to so.

jeepdrvr71, 2020-07-03:
That did the trick. Maybe that is something you should add to the instructions. Delete your old config.json and rerun ./nhl_setup

jeepdrvr71, 2020-07-19:
everything working great

closed 2020-07-19 by jeepdrvr71