riffnshred/nhl-led-scoreboard #110: Error when running sudo Python3

dawelge01, 2020-07-02:
Traceback (most recent call last):
  File "src/main.py", line 12, in <module>
    from api.weather.ecWeather import ecWxWorker
  File "/home/pi/nhl-led-scoreboard/src/api/weather/ecWeather.py", line 1, in <module>
    from env_canada import ECData
  File "/usr/local/lib/python3.7/dist-packages/env_canada/__init__.py", line 2, in <module>
    from .ec_radar import *
  File "/usr/local/lib/python3.7/dist-packages/env_canada/ec_radar.py", line 38, in <module>
    font = ImageFont.truetype('Arial.ttf', size=40)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 280, in truetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 145, in __init__
    layout_engine=layout_engine)
OSError: cannot open resource

**Logs**
Copy and past any error or messages that you can get. 

**Your config file**
A screen shot of your config.json file can be really useful to help you. 

**Hardware you are using**
- Raspberry pi model Raspberry pi zero wh
- The Matrix driver (Adafruit Bonnet or Adafruit HAT) Adafruit Bonnet 
- Anti-flicker mod or not? Not 

**Software**
The latest, just installed.

Tanosh11, 2020-07-02:
I had this same issue a few days ago. I don't know the full technicalities (are you on a Mac by chance?), but this resolved it for me:

https://stackoverflow.com/questions/47694421/pil-issue-oserror-cannot-open-resource?rq=1

Installed the arial font linked and everything worked after that.

dawelge01, 2020-07-02:
I'm not on a MAC. Is this after you did the configuration setup? I was trying to run the board to test and it was a no go. I get the error.

Tanosh11, 2020-07-02:
Yes, I did. My apologies, I did this a few days ago and forgot. I also ran this to install the Win fonts on the pi:

sudo apt-get install ttf-mscorefonts-installer

dawelge01, 2020-07-02:
Great that worked!! I have a clock displayed how do I get the weather to show up?

Thanks-

Current Message
INFO (12:05:41): Checking for new day
INFO (12:05:41): It is not a new day
INFO (12:05:47): ['07/02 12:05 PM', '\uf00d', 'Sunny', '28.8C', '33.8C', '48%', '17.0C']
INFO (12:05:47): ['15.0 kph', 'E', '\uf061', '0.0 kph', '1017.0 hPa', '\uf07b', '16.1 km']
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/nhl-led-scoreboard/src/api/weather/ecAlerts.py", line 76, in run
    warn_datetime = datetime.datetime.strptime(warn_date,self.alert_date_format)
  File "/usr/lib/python3.7/_strptime.py", line 577, in _strptime_datetime
    tt, fraction, gmtoff_fraction = _strptime(data_string, format)
  File "/usr/lib/python3.7/_strptime.py", line 359, in _strptime
    (data_string, format))
ValueError: time data 'Thursday July 02, 2020 at 09:29 EDT' does not match format '%A %B %d, %Y at %H:%M %Z'

INFO (12:06:03): Checking for new day
INFO (12:06:03): It is not a new day
INFO (12:06:23): Checking for new day
INFO (12:06:23): It is not a new day
INFO (12:06:42): Checking for new day
INFO (12:06:42): It is not a new day

Tanosh11, 2020-07-02:
That's actually where I'm stuck too. I can get all the boards "working" except weather. Depending on where you (Canada/US) you may need an API key from here: https://openweathermap.org/api

I still can't get mine to update though. I should probably submit a separate issue ¯\_(ツ)_/¯

dawelge01, 2020-07-02:
Is it free or do you have to pay for it?

Tanosh11, 2020-07-02:
It's free. There are paid tiers listed, you need to create an account to get started.

dawelge01, 2020-07-02:
Which did you choose?

Tanosh11, 2020-07-02:
I chose the free tier and put my key in via the manual setup. Like I said though, I'm still having issues with it. If you figure it out let me know.

dawelge01, 2020-07-02:
Looks like there might be issue with it: https://github.com/riffnshred/nhl-led-scoreboard/issues/109

riffnshred, 2020-07-02:
we found the bug last night, just need to do some test after work before I upload the update. Sorry for the hassle.

Tanosh11, 2020-07-02:
Whoa no hassle at all, this has been a fun project! Thank you for everything you've done!

riffnshred, 2020-07-02:
Thanks !!! Not done with it, can't wait for hockey to comeback 8)

dawelge01, 2020-07-02:
Yes thanks for everything! This is a neat project. I have one other questions how do you display other screens? Say I wanted Standings, Clock and COVID. Thanks again!

riffnshred, 2020-07-02:
Thanks! In the config.json on that off_day state, put 

`"standings", "clock", "covid_19"`

you can also do it through the ./nhl_setup, just make you say no fast one team config setup at the beginning to go through all the options. when it will come to selecting options for the off_day state, pick up the "standing" board and then it will ask you if you want to add an other one for the off_day state. say yes and then pick the next one you want, rinse and repeat.

dawelge01, 2020-07-02:
I figure that's what it was I wanted to make sure. Again I can't wait for hockey to start! Would like to see this thing in Action! I mention something to someone else on another site, it would be cool to connect a bluetooth speaker and have a goal horn go off. Just a thought for future updates.

riffnshred, 2020-07-03:
I just pushed an update that should have fixed this. could you confirm @dawelge01 

Also, some guys on discord are working on something similar using smart home stuff. The issue with using bluetooth is that we recommend deactivating it for reduce the flickering issue of the matrix. There are other ways todo something like that tho for sure.

Tanosh11, 2020-07-06:
Not OP, but since I posted here before I can confirm the weather API is now pulling properly. It has my IP halfway across the state (I have it at work), so I'll override that. Thanks again!

dawelge01, 2020-07-07:
Well I was going to mess with it but my Raspberry Pi W stopped working on the WiFi. Any suggestions?

falkyre, 2020-07-07:
@Tanosh11 the way the IP address lookup works is where your service provider has registered the IP. It doesn't always work precisely which is why putting in your city and state or city and province is offered. If that is still an issues, I was thinking of adding in the latitude and longitude as an option in the config but so far haven't seen a need for it.

Tanosh11, 2020-07-07:
@falkyre Yeah since it's at work our IP is routing through a bunch of offsite locations. I looked up my city on openweather and put in the same naming convention they had. I didn't use the state, I used "CityName", US. That seemed to work too, all the data is accurate. It works really well, makes the display even better, thank you!

dawelge01, 2020-07-07:
@Tanosh11 it shows weather just not sure where it is coming from. It's hot here I'm at lol

dawelge01, 2020-07-07:
@riffnshred The Temp is correct. It has hard to understand because there is no location of where you are. I had to double check between OWN and the board. This would be a great add to show the City. Thanks for your help! So now we have to wait until hockey starts to see if the score board works!

riffnshred, 2020-07-07:
Glad you got it working bud. The issue with showing the City is that there is not much space. Maybe someone could do a more detailed oriented board in the future with smaller size text and no image.

closed 2020-07-07 by riffnshred