riffnshred/nhl-led-scoreboard #111: config.json valitation errors

riffnshred, 2020-07-03:
**Describe the bug**
Scoreboard return config.json error in the log and wont work.

**Logs**
: Now validating config.json.....
ERROR (20:48:45): config.json fails validation: error: [data.boards.covid19.country must be array]
ERROR (20:48:45): Rerun the nhl_setup app to create a valid config.json
INFO (20:48:49): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
INFO (20:48:49): Now validating config.json.....
ERROR (20:48:50): config.json fails validation: error: [data.boards.covid19.country must be array]
ERROR (20:48:50): Rerun the nhl_setup app to create a valid config.json
INFO (20:48:54): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
INFO (20:48:54): Now validating config.json.....
ERROR (20:48:54): config.json fails validation: error: [data.boards.covid19.country must be array]
ERROR (20:48:54): Rerun the nhl_setup app to create a valid config.json
INFO (20:48:58): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
INFO (20:48:58): Now validating config.json.....
ERROR (20:48:58): config.json fails validation: error: [data.boards.covid19.country must be array]
ERROR (20:48:58): Rerun the nhl_setup app to create a valid config.json

This is due to nhl_setup putting `null` instead of square bracket `[]` when the user select to not show country stats or canada proving or us states.

riffnshred, 2020-07-03:
Patial solution is to replace the `null` for `[]` instead. 

Exemple `"country"=null, ` would become `"country=[]"`

falkyre, 2020-07-03:
For anyone getting these errors and have used the nhl_setup to create your config.json, what are you selecting for the covid boards in the nhl_setup?

falkyre, 2020-07-03:
Also for anyone getting this, do you already have an existing config.json when you run the nhl_setup?

riffnshred, 2021-01-04:
I believe this was fixed or just a compatibility problem between config files of different versions.

closed 2021-01-04 by riffnshred