riffnshred/nhl-led-scoreboard #112: Distorted image on rgb-led-matrix

saxonpaul92, 2020-07-06:
**Describe the bug**
Installed the nhl-led-scoreboard from [Riffnshreds github page](https://github.com/riffnshred/nhl-led-scoreboard
)...  

Hardware:
Raspberry Pi Zero
16GB SD card
Adafruit-led-matrix bonnet with anti-flicker mod
32x64 LED Matrix ([DFRobot ](https://www.dfrobot.com/product-1569.html)P4 matrix)

**Logs**
Logs are standard output... no errors...
INFO (17:31:12): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
INFO (17:31:12): Now validating config.json.....
INFO (17:31:17): config.json passes validation
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
INFO (17:31:20): location is: Winnipeg, MB [49.8955367, -97.1384584]
DEBUG (17:31:21): No next game detected for Devils
DEBUG (17:31:21): No next game detected for Islanders
DEBUG (17:31:21): No next game detected for Rangers
DEBUG (17:31:21): No next game detected for Flyers
DEBUG (17:31:21): No next game detected for Penguins
DEBUG (17:31:21): No next game detected for Bruins
DEBUG (17:31:21): No next game detected for Sabres
DEBUG (17:31:21): No next game detected for Canadiens
DEBUG (17:31:21): No next game detected for Senators
DEBUG (17:31:21): No next game detected for Maple Leafs
DEBUG (17:31:21): No next game detected for Hurricanes
DEBUG (17:31:21): No next game detected for Panthers
DEBUG (17:31:21): No next game detected for Lightning
DEBUG (17:31:21): No next game detected for Capitals
DEBUG (17:31:21): No next game detected for Blackhawks
DEBUG (17:31:21): No next game detected for Red Wings
DEBUG (17:31:21): No next game detected for Predators
DEBUG (17:31:21): No next game detected for Blues
DEBUG (17:31:21): No next game detected for Flames
DEBUG (17:31:21): No next game detected for Avalanche
DEBUG (17:31:21): No next game detected for Oilers
DEBUG (17:31:21): No next game detected for Canucks
DEBUG (17:31:21): No next game detected for Ducks
DEBUG (17:31:21): No next game detected for Stars
DEBUG (17:31:21): No next game detected for Kings
DEBUG (17:31:21): No next game detected for Sharks
DEBUG (17:31:21): No next game detected for Blue Jackets
DEBUG (17:31:21): No next game detected for Wild
DEBUG (17:31:21): No next game detected for Jets
DEBUG (17:31:21): No next game detected for Coyotes
DEBUG (17:31:21): No next game detected for Golden Knights
INFO (17:31:24): /sbin/reboot <-- reboot || poweroff --> /sbin/poweroff
INFO (17:31:25): Refreshing EC current observations weather
INFO (17:31:25): Checking for EC weather alerts
INFO (17:31:41): Checking update status for /home/pi/nhl-led-scoreboard...
INFO (17:31:55): Code up to date.
Exiting NHL-LED-SCOREBOARD

**Your config file**
{
    "debug": true,
    "live_mode": false,
    "preferences": {
        "time_format": "24h",
        "end_of_day": "8:00",
        "location": "Winnipeg, MB",
        "live_game_refresh_rate": 10,
        "teams": [
            "Stars"
        ],
        "sog_display_frequency": 4
    },
    "goal_animations": {
        "pref_team_only": true
    },
    "states": {
        "off_day": [
            "covid_19",
            "clock"
        ],
        "scheduled": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock"
        ],
        "intermission": [
            "scoreticker"
        ],
        "post_game": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "wild_card",
            "divisions": "central",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false
        },
        "covid19": {
            "worldwide_enabled": false,
            "country_enabled": true,
            "country": [
                "USA"
            ],
            "us_state_enabled": false,
            "us_state": [
                "New York"
            ],
            "canada_enabled": false,
            "canada_prov": [
                "Quebec"
            ]
        },
        "weather": {
            "enabled": true,
            "units": "metric",
            "duration": 60,
            "data_feed": "EC",
            "alert_feed": "EC",
            "owm_apikey": "",
            "update_freq": 5,
            "show_alerts": true,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "dimmer": {
            "enabled": false,
            "source": "software",
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": true,
            "bonnet": true,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
} 

**Software**
V.1.0.0
Raspbian Release 5/27/2020

saxonpaul92, 2020-07-07:
Follow-up... 

It appears that the 32x64 led matrix is only working on the second and fourth rows of 8 leds (rows 0-7 = off) (row 8-15 = on) (row 16 - 23 = off) and (row 24 - 31 = on)....   the matrix is printing all of the information for rows 0-31 in row 8-15 and 24-32).

That is causing the distortion.

I have tried every flag with no luck.  It almost certainly a driver issue.

Tanosh11, 2020-07-07:
What's the command you're using to start the display?

You've tried some of the flags listed under "GPIO Speed" and "Panel Connection" here?:  https://github.com/hzeller/rpi-rgb-led-matrix

Do any of the other demos work? I had something similar happen and once I oriented the display and slowed the speed it worked properly.

saxonpaul92, 2020-07-07:
I have tried many command option flags.  

sudo python3 runtext.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat-pwm --led-brightness=50 --led-scan-mode=1 --led-multiplexing =0 --led-chain=1 --led-parallel=1 --led-pwm-bits=11 --led-pwm-lsb=130 --led-show-refresh --led-slowdown-gpio=2 --led-no-hardware-pulse=1 --led-rgb-sequence=RGB --led-row-addr-type=0

it is overkill but, I use this to vary the parameters to try and get a combination that works.

At present, I am running the runtext.py demo and I have it scrolling "Hello World" and is readable however, rows 0-7 and 16-23, do not work (blank) and the dots that are addressed in those areas are mirrored back to the active rows.

It seems to me that this must be an addressing issue as it appears that everything is rendered to the led matrix but is bunched up in the active rows.  This means that these addresses are getting written to twice, which distorts the display.

riffnshred, 2020-07-08:
You said on Discord that you had it running correctly before and had a crash so you had to reinstall everything. Do you have any idea what cause the crash. That's the kind of issue that you might find an answer on the [rgbmatrix repo](https://github.com/hzeller/rpi-rgb-led-matrix/issues).

saxonpaul92, 2020-07-08:
SOLVED:

After much testing and configuring I finally came upon the problem.

The Raspberry Pi Zero has a 4-pin header that sits beside the 40-pin GPIO header.

The 4-pin header is used for Composite Video, power switching, etc...

When the Adafruit RGB matrix bonnet is installed it touches the top of those pins and causes a short between the bonnet and the Pi.

This short caused the problem of banding on the display and most likely contributed to crashing and breaking the RPi.

I trimmed down the header and used another RPi 0 and problem solved.

One other issue I noticed is the RPi 0 is really working hard with this project.

[image]

You can see from htop that the RPi 0 is normally staying between 85-100% CPU utilization in while running the NHL-scoreboard.

For most of my projects I really don't like to go over 80% average utilization with spikes into the 81-100% range.  

Maybe some optimizing of the code might help or just not using the RPi.

Thanks for the comments and hopefully this will help someone else.

falkyre, 2020-07-09:
There's not much that can really be done with the code due to a few things with how the RGB matrix library is created (the creator of that library has tried optimizing it already) plus how Python works so we already come up against the wall with that.   We have had discussions for a few things for the next major version but haven't settled on exactly what will be done to try and improve performance. 
 The better base rpi model is the 3A+.  It's quad core so the CPU utilization is still manageable (although one core will always be hovering around 80-100% usage depending on what's happening.  Scrolling is hard on the CPU).

riffnshred, 2020-07-09:
@saxonpaul92  Glad you found the issue. 

Like falkyre said, We've been developing with the Pi Zero as minimum requirement so far. As we add new features we try to keep it that way but for what I have in mind for the future, It probably wont be powerful enough. That being said, my intention is to keep V1 active and to make new features still. It just wont have some of the more demanding stuff V2 will provide.

closed 2020-07-09 by riffnshred

saxonpaul92, 2020-07-10:
I switched over to the RPi 2 and the CPU is sitting very nicely at 68%.

filifunk, 2020-07-16:
@saxonpaul92 Hi, I think I had the same problem as you.  I made sure the adafruit hat wasn't touching the 4 pin and it looked different (but not fixed)

I'm going to come off like a noob, because I am.  How did you find out what that 4 pin header was for?  Did you look at the schematic here:  https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_Zero_1p3_reduced.pdf

I have a hard time finding it on there.

Also once the short happens...you used a different raspberry pi.  Is the old one ruined for this project?  Why am I still able to turn on the raspberry pi and it looks like its functioning perfectly normal?

How did you trim down the header?

Thanks, apologize I'm such a noob.

saxonpaul92, 2020-07-17:
Yes, the schematic shows it but you can also look on the RPi Zero and 2 of
the pins are marked "RUN".  That is for a reset switch.  The other 2 pins
are marked "TV".  That is for composite video.

I changed out the board to a RPi 2 because the single core processor of the
RPi is really getting overworked.  The RPi 2 or later works alot better.