riffnshred/nhl-led-scoreboard #113: ImportError: libjasper.so.1 when first testing

tvarty97, 2020-07-07:
Followed the guide and everything was working fine (including the hello world test) up until the "Usage" section when you first test before setting it up to run on its own.

Logs:
:~/nhl-led-scoreboard $ sudo python3 src/main.py --led-gpio-mapping=adafruit-hat --led brightness=60 --led-slowdown-gpio=2
Traceback (most recent call last):
  File "src/main.py", line 12, in <module> from api.weather.ecWeather import ecWxWorker
  File "/home/pi/nhl-led-scoreboard/src/api/weather/ecWeather.py", line 1, in <module> from env_canada import ECData
  File "/usr/local/lib/python3.7/dist-packages/env_canada/__init__.py", line 2, in <module>from .ec_radar import *
  File "/usr/local/lib/python3.7/dist-packages/env_canada/ec_radar.py", line 9, in <module> import cv2
  File "/usr/local/lib/python3.7/dist-packages/cv2/__init__.py", line 3, in <module> from .cv2 import *
ImportError: libjasper.so.1: cannot open shared object file: No such file or directory

I'm using a Zero with the bonnet without the flicker mod and the most recent software (first attempt and downloaded a few days ago).

riffnshred, 2020-07-07:
If got V1.2.1 (released last weekend) installed or do you got a previous version? Id suggest rerunning the install.sh and see if that fix it, otherwise a complet intall from scratch will definitly fix that.

falkyre, 2020-07-08:
Oh man, looks like there was another env Canada library release. The guy that wrote it is using opencv it looks like.  @riffnshred I think the install.sh needs to force the env Canada version I have in the requirements.txt

falkyre, 2020-07-08:
Quick fix

falkyre, 2020-07-08:
sudo pip3 uninstall env-canada

falkyre, 2020-07-08:
sudo pip3 install env-canada==0.0.35

riffnshred, 2020-07-08:
Hah what the hell ??? Thanks I'll fix it and push an update later tonight.

riffnshred, 2020-08-08:
This was fixed with v1.3.0

closed 2020-08-08 by riffnshred