riffnshred/nhl-led-scoreboard #114: Can't run nhl setup

filifunk, 2020-07-11:
I'm trying to run nhl_setup and it won't work.

This is the error I get:
`bash: .: nhl_setup: cannot execute binary file`

Any idea what I should do?

I'm using an Adafruit Bonnet I think.

I don't know how to check what version I'm using.

Clearly I am a noob!

riffnshred, 2020-07-11:
Were you able to run the sample script first? That's a weird one. Make sure you are in the nhl-led-scoreboard root folder when you run it. from anywhere just do `cd` then `cd nhl-led-scoreboard` and then run `./nhl_setup`

filifunk, 2020-07-12:
I was able to run the sample script!  What showed up on my LED was unintelligible, but it worked.  

Just tried it again and I think I was adding an extra space in between the . and the /.  But now I get a new error:

`[2062] Error loading Python lib '/tmp/_MEIGKKgHK/libpython3.7m.so.1.0': dlopen: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/_MEIGKKgHK/libpython3.7m.so.1.0)
`

Kind of weird that it would want something in a temporary file?

filifunk, 2020-07-14:
Saw that some people include their config file...my config file is just the sample config:

{
	"debug": true,
	"live_mode": false,
	"preferences": {
		"time_format": "24h",
		"end_of_day": "8:00",
		"location": "Winnipeg, MB",
		"live_game_refresh_rate": 10,
		"teams": [
			"Flames",
			"Oilers"
		],
		"sog_display_frequency": 4
	},

	"goal_animations": {
		"pref_team_only": true
	},

	"states":{
		"off_day": [
			"covid_19",
			"clock"
			
		],
		"scheduled": [
			"team_summary",
			"standings",
			"scoreticker",
			"clock"
		],
		"intermission": [
			"scoreticker"
		],
		"post_game": [
			"team_summary",
			"standings",
			"scoreticker",
			"clock"
		]
	},

	"boards": {
		"scoreticker": {
			"preferred_teams_only": false,
			"rotation_rate": 5
		},
		"standings": {
			"preferred_standings_only": true,
			"standing_type": "wild_card",
			"divisions": "central",
			"conference": "eastern"
		},
		"clock": {
			"duration": 15,
			"hide_indicator": false
		},
        "covid19": {
            "worldwide_enabled": false,
            "country_enabled": true,
            "country": [
                "USA"
            ],
            "us_state_enabled": false,
            "us_state": [
                "New York"
            ],
            "canada_enabled": false,
            "canada_prov": [
                "Quebec"
            ]
		},
		"weather": {
			"enabled": true,
			"units": "metric",
			"duration": 60,
			"data_feed": "EC",
			"alert_feed": "EC",
			"owm_apikey": "",
			"update_freq": 5,
			"show_alerts": true,
			"alert_title": true,
			"scroll_alert": true,
			"alert_duration": 5,
			"show_on_clock": true
		}	
	},
	"sbio": {
		"dimmer": {
			"enabled": false,
			"source": "software",
			"frequency": 5,
			"light_level_lux": 400,
			"mode": "always",
			"sunset_brightness": 5,
			"sunrise_brightness": 40
		},
		"pushbutton": {
			"enabled": true,
			"bonnet": true,
			"pin": 25,
			"reboot_duration": 2,
			"reboot_override_process": "",
			"display_reboot": true,
			"poweroff_duration": 10,
			"poweroff_override_process": "",
			"display_halt": true,
			"state_triggered1": "weather",
			"state_triggered1_process": ""
		}
	}
}

falkyre, 2020-07-15:
The error with loading glib typically is a corrupt SD card.  The nhl_setup is a compiled python script and will uncompress files as needed that's why you are seeing stuff in /tmp.

filifunk, 2020-07-15:
That was totally it @falkyre , thank you!

closed 2020-07-15 by filifunk

falkyre, 2020-07-15:
@filifunk no problem. Glad you got it going.