riffnshred/nhl-led-scoreboard #116: Board for betting ticker

blaineA, 2020-07-17:
Was wondering if there is a way to get the over/under and other standard betting results on a board. For example: Pens -110 Flyers +110 o/u 5.5
Could also have every team on there a rolling ticker?

Juts something I was thinking about as sports open back up

joshkay, 2020-07-17:
Pretty sure that this isn't available via the public NHL api.  May be possible with a separate API, but likely not free.

riffnshred, 2020-07-17:
What @joshkay said. Im sure there is a way to grab the data by web scrapping or something. Someone could definitly fork the repo and make it but thats something I don't think many would use. I might be wrong tho.

machinasdei, 2020-11-17:
I'd like to do this as well, and perhaps add other APIs inserted to run in the offseason (aside from the covid19) has anyone successfully done this yet?  I suck at scripting, but would love to do it.

riffnshred, 2020-11-17:
At this point in the project, Im not adding anything else thats not directly nhl related. I Got an other app in the making that will be made for running different apps for rpi rgb matrix. That will come later after Im done with the next version of the nhl-led-scoreboard.

closed 2020-11-17 by riffnshred