riffnshred/nhl-led-scoreboard #117: Can't run the sample runtext.py or the regular full version

filifunk, 2020-07-18:
**Describe the bug**

I am trying to run the sample runtext.py here:

https://youtu.be/zj5DPYMYXgw

I haven't been able to get anything to legibly show up on the LED screen.

I have reformatted my SD card and put Raspbian on it again.  It still doesn't work.
I thought my problem was like another person who has posted where the adafruit hat may have been touching other pins.  I made sure they weren't.  Looking at where the pins touched my adafruit hat it looked like it didn't cause any damage or was touching anything that may have been damaged.  It still doesn't work.

**Logs**
I don't get any errors but this is what shows up when I run the runtext.py

sudo python3 runtext.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-brightness=60
Press CTRL-C to stop sample

pretty normal.

When I try to run the regular version:

sudo python3 src/main.py --led-gpio-mapping=adafruit-hat --led-brightness=60 --led-slowdown-gpio=2
INFO (10:59:22): NHL-LED-SCOREBOARD - v1.2.1 (64x32)
INFO (10:59:22): Now validating config.json.....
INFO (10:59:23): config.json passes validation
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
INFO (10:59:23): location is: Winnipeg, MB [49.8955367, -97.1384584]
DEBUG (10:59:23): No next game detected for Devils
DEBUG (10:59:23): No next game detected for Sabres
DEBUG (10:59:23): No next game detected for Senators
DEBUG (10:59:23): No next game detected for Red Wings
DEBUG (10:59:23): No next game detected for Ducks
DEBUG (10:59:23): No next game detected for Kings
DEBUG (10:59:23): No next game detected for Sharks
INFO (10:59:23): /sbin/reboot <-- reboot || poweroff --> /sbin/poweroff
INFO (10:59:23): Refreshing EC current observations weather
INFO (10:59:23): Checking for EC weather alerts
INFO (10:59:23): Checking update status for /home/pi/nhl-led-scoreboard...
INFO (10:59:25): Code up to date.
INFO (10:59:26): Rendering...
DEBUG (10:59:26): refresing data
INFO (10:59:26): Live mode is off. Going through the boards
DEBUG (10:59:26): PING !!! Render off day
INFO (10:59:27): Checking for new day
INFO (10:59:27): It is not a new day
DEBUG (10:59:27): refresing data
DEBUG (10:59:27): Grabbing all COVID-19 data from API
INFO (10:59:29): No active EC weather alerts in your area
INFO (10:59:29): ['07/18 10:59', '\uf07b', 'Curr Cond N/A', '20.4C', '24.5C', '73%', '15.5C']
INFO (10:59:29): ['9.0 kph', 'SW', '\uf05a', '0.0 kph', '1005.0 hPa', '\uf057', '24.1 km']

**Your config file**

{
    "debug": true,
    "live_mode": false,
    "preferences": {
        "time_format": "24h",
        "end_of_day": "8:00",
        "location": "Winnipeg, MB",
        "live_game_refresh_rate": 10,
        "teams": [
            "Blackhawks"
        ],
        "sog_display_frequency": 4
    },
    "goal_animations": {
        "pref_team_only": true
    },
    "states": {
        "off_day": [
            "covid_19",
            "clock"
        ],
        "scheduled": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock"
        ],
        "intermission": [
            "scoreticker"
        ],
        "post_game": [
            "team_summary",
            "standings",
            "scoreticker",
            "clock"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "wild_card",
            "divisions": "central",
            "conference": "eastern"
        },
        "clock": {
            "duration": 15,
            "hide_indicator": false
        },
        "covid19": {
            "worldwide_enabled": false,
            "country_enabled": true,
            "country": [
                "USA"
            ],
            "us_state_enabled": false,
            "us_state": [
                "New York"
            ],
            "canada_enabled": false,
            "canada_prov": [
                "Quebec"
            ]
        },
        "weather": {
            "enabled": true,
            "units": "metric",
            "duration": 60,
            "data_feed": "EC",
            "alert_feed": "EC",
            "owm_apikey": "",
            "update_freq": 5,
            "show_alerts": true,
            "alert_title": true,
            "scroll_alert": true,
            "alert_duration": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "dimmer": {
            "enabled": false,
            "source": "software",
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": true,
            "bonnet": true,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
}

**Hardware you are using**
- Raspberry pi 3b+
- Adafruit RGB Matrix Bonnet for Raspberry Pi
- no antiflicker mod 

**Software**
- 1.2.1.

I noticed when I run the program it says: 

INFO (10:59:22): NHL-LED-SCOREBOARD - v1.2.1 (64x32)

Will 64x32 work with my 32 rows 64 column (32x64) LED board?  Sometimes I feel like the lights aren't put on the board correctly.

falkyre, 2020-07-18:
Where did you buy the RGB matrix board?  It almost sounds like you have a bad data cable or it's a board that doesn't work out of the box with the matrix library (which is not written by us, we just use it).  Have you gone through the troubleshooting section of the matrix library?

falkyre, 2020-07-18:
https://github.com/hzeller/rpi-rgb-led-matrix

filifunk, 2020-07-18:
I got the RGB matrix board from Arrow.com, it's an adafruit 2277 board.  Which is the same board that was used in the tutorial I saw that brought me to this project: https://www.youtube.com/watch?v=EPZawKPC73k

Yeah I was trying all of that last night for the most part.  Going off of those troubleshooting options:

> Do not use a graphical user interface (Even though the Raspberry Pi foundation makes you believe that you can do that: don't. Using a Pi with a GUI is a frustratingly slow use of an otherwise perfectly good embedded device.)

_- I flashed the version of raspbian with less software.  I suppose I could try the lightest version but not sure that would make the difference here._

> Switch off on-board sound (dtparam=audio=off in /boot/config.txt). External USB sound adapters work, and are much better quality anyway, so that is recommended if you happen to need sound. The on-board sound uses a timing circuit that the RGB-Matrix needs (it seems in some distributions, such as arch-linux, this is not enough and you need to explicitly blacklist the snd_bcm2835 module).

_- did this!_

> Don't run anything that messes in parallel with the GPIO pins, e.g. PiGPIO library/daemon or devices that use the i2c or 1-wire interface if they are on the same pins you need for the panel.

_- I am using a pijuice battery hat.  I took that off and just attached the Matrix bonnet onto the raspberry pi and had the same results.  The pijuice battery hat actually is more convenient because it is built in a way that avoids anything from touching any of the other pins._

> I have also seen reports that on some Pis, the one-wire protocol is enabled (w1-gpio). This will also not work (disable by removing dtoverlay=w1-gpio in /boot/config.txt; or using raspi-config, Interface Options -> 1-Wire)

_- I don't have the dtoverlay=w1-gpio in my config file to begin with so didn't do anything here._

> If you see some regular flickering, make sure that there is no other process running on the system that could cause that. For instance, it is known that merely running top creates a faint flicker every second it updates. Or a regular ntp run can also cause flicker once a minute (switch off with sudo timedatectl set-ntp false). Maybe instead you might want to run ntp at system start-up but then not regularly updating. There might be other things running regularly you don't need; consider a sudo systemctl stop cron for instance. To address some irregular flicker, consider the --led-limit-refresh option.

_- I don't think flickering is my problem_

> There are probably other processes that are running that you don't need and remove them; I usually remove right away stuff I really don't need e.g.
> 
> sudo apt-get remove bluez bluez-firmware pi-bluetooth triggerhappy pigpio
> Take a close look at your systemd (systemctl) and see if there are other things running you don't need. If you have seen packages in standard Raspbians that interfere with the matrix code, let me know to include it here. In general: This is why starting with a minimal installation is a good idea: there is simply less cruft that you have to disable.

_- I don't think I know enough about computers to know what I should shut off in systemctl_

> It seems that more recent version of Raspbian Lite result in some faint brightness fluctuations of the displays and it is not quite clear why (see issue #483). If you are a Kernel person and can help figuring out what is happening that would be very appreciated. Also, you might know a minimal Linux distribution that is more suited for near realtime applications ?

_- brightness fluctuations aren't really my issue_

Would you expect that I shorted something?  I wish I could think of a way to pinpoint exactly where the problem is.  If it is a bad data cable, is there something I could do to find out for sure?

filifunk, 2020-07-18:
Nevermind, I figured it out.  You saying it looked like a hardware problem made me realized that I wasn't powering the Adafruit bonnet with a separate 5V plug.  I figured I would just have my pi plugged in and the battery and it would be fine.  And it did show some lights on the LED so I thought it was fine.  I just left my battery on and plugged in the RGB bonnet and it works...I feel like an idiot.  Sorry!!!!

closed 2020-07-18 by filifunk