riffnshred/nhl-led-scoreboard #118: Weather Warning Issue

jeepdrvr71, 2020-07-19:
The time is over the weather warning screen. Is there a way to turn the time off on this screen.

[image]

riffnshred, 2020-07-19:
I'll add that for future bug fix but for now go under config/layout/layout.json, scroll down to wx_alert and change the font of "advisory_ca" (for canada) or "advisory_us" (for US) from "wx_alert" to "wx_medium". If you choose set "scroll_alert" to true under weather in the config.json, the font will be at medium instead of the bigger size you currently see.

falkyre, 2020-07-22:
This will be fixed in the next weather release.  Hopefully next week sometime now that I'm back from vacation

riffnshred, 2020-08-08:
I believe this is fixed and can be closed @falkyre ?

falkyre, 2020-08-09:
Yup. It's fixed. You can close it off

closed 2020-08-09 by riffnshred