riffnshred/nhl-led-scoreboard #122: Switch to scoreboard after favorite team game

Pending31, 2020-08-02:
(First, thanks; this board is awesome). 
Now for my issue/suggestion...
I Have the post-game set to scoreboard, yet after a favorite team game, it stays on that game showing the final. How would I go about changing this to go back to the league scoreboard?  (I’m relatively new at Python, though know enough about VB and C++ that it seems like it’s doable)

cmcurtis33, 2020-08-02:
Was experiencing this same issue, and it's more prevalent now that the games are scheduled throughout the day such that if your favorite team plays early then you are looking at the final score the rest of the day. Here's what I did to make the behavior as you are requesting:

Within folder nhl-led-scoreboard/src/renderer, edit the main.py file. Strongly suggest you make a backup of the file first (can use command 'cp main.py main.py.backup')

Edit the main.py file, and scroll down to around line 115. You will see an 'elif' statement with next line being debug output of "Game Over". At the bottom of this block (right after the self.sleep Event.wait() call, add the following:

`                if self.data._next_game():
                    debug.info("moving to the next preferred game")
                    return
                self.boards._post_game(self.data, self.matrix,self.sleepEvent)
`

the full 'elif' statement is shown below after this modification:

`            elif self.status.is_game_over(self.data.overview.status):
                debug.info("Game Over")
                self.scoreboard = Scoreboard(self.data.overview, self.data)
                self.__render_postgame(self.scoreboard)
                # sleep(self.refresh_rate)
                self.sleepEvent.wait(self.refresh_rate)
                if self.data._next_game():
                    debug.info("moving to the next preferred game")
                    return
                self.boards._post_game(self.data, self.matrix,self.sleepEvent)
`

Hope this helps. And.. want to be able to propose this mod to the main source in Git/GitHub. Still working on that as I'm new in this source code tool.

cmcurtis33, 2020-08-02:
Was experiencing this same issue, and it's more prevalent now that the games are scheduled throughout the day such that if your favorite team plays early then you are looking at the final score the rest of the day. Here's what I did to make the behavior as you are requesting:

Within folder nhl-led-scoreboard/src/renderer, edit the main.py file. Strongly suggest you make a backup of the file first (can use command 'cp main.py main.py.backup')

Edit the main.py file, and scroll down to around line 115. You will see an 'elif' statement with next line being debug output of "Game Over". At the bottom of this block (right after the self.sleep Event.wait() call, add the following:

```
if self.data._next_game(): 
  debug.info("moving to the next preferred game") 
  return
self.boards._post_game(self.data, self.matrix,self.sleepEvent)
```

the full 'elif' statement is shown below after this modification:

```
elif self.status.is_game_over(self.data.overview.status):
  debug.info("Game Over") 
  self.scoreboard = Scoreboard(self.data.overview, self.data) 
  self.__render_postgame(self.scoreboard) 
  # sleep(self.refresh_rate) 
  self.sleepEvent.wait(self.refresh_rate) 
  if self.data._next_game(): 
    debug.info("moving to the next preferred game") 
    return 
  self.boards._post_game(self.data, self.matrix,self.sleepEvent)
```

Hope this helps. And.. want to be able to propose this mod to the main source in Git/GitHub. Still working on that as I'm new in this source code tool.

riffnshred, 2020-08-02:
It is supposed to go back to the post game state which is showing all the boards you selected for that state, but last week for the exhibition games the "Final" state code was different and because of that the scoreboard was staying into a "Game Over" status. I got a beta version up and running that I changed how to interpret that specific status code but then I looked at the official playoff game last night and now it's all back to normal. 

The one issue right now with the _next_game function is not working properly and the board wont move to the NEXT preferred game but will stay into post game state showing the boards you selected for that state. that is fixed in the beta and will be release either Tuesday or Wednesday.

cmcurtis33, 2020-08-02:
@riffnshred Excellent.. looking forward to the next release. Excited we have hockey back again.

Pending31, 2020-08-02:
Is it programmed to update automatically?  Maybe it's the round-robin playoff format, but it's still showing the final screen after today's Bruins game.  (Or it just dislikes Brad Marchand)  I'll try editing the  main file.

Non-related... any plans to add series scoreboards once we get into the full playoffs?

riffnshred, 2020-08-03:
At what time is you `end_of_day` set? I suggest keeping it in the morning like between 00:01 and 12:00. Its a bit weird how this works but its because the API update the schedule at Noon and to show the new schedule earlier, I use this setting to if its morning and I set it to like 8:00, it will show me the schedule of the current day before the API update. if its set after noon, the API will switch to the current day AFTER that set time. so lets say you put it at 3 in the afternoon, the scoreboard will show the result of the previous day until 3 pm.

As for you unrelated question....

https://streamable.com/pz2jh2

Pending31, 2020-08-03:
Thanks.  I had the day change set at 4am To allow west coast games to finish before showing the next day.   I did try the edit suggested above, but didn’t render anything after reboot, so I went and restored the original to get it working. (Still need to go and double check I didn’t fat-finger the edit, as it was late when. I was doing this)

Pending31, 2020-08-04:
Just confirmed, it's still not updating.  I reconfigured to a different favorite team (hurricanes instead of bruins; figuring they're not in the round-robin), and it stayed on the final.  I went back and went to edit the main.py file as indicated above, (and made sure it was correct) the screen doesn't render at all.  Restoring back to the original works.  Any ideas?

cmcurtis33, 2020-08-04:
@Pending31 The next version should be released soon.. but if you want to try beforehand, attached is the main.py from my src\renderer with the modification. If this still doesn't work, it could be your config does not have a setting enabled. Again, backup your main.py before overwriting. Hope this helps.

[main.py.zip](https://github.com/riffnshred/nhl-led-scoreboard/files/5019747/main.py.zip)

Pending31, 2020-08-05:
Thanks, this worked like a charm and it's working fine now.  I'll go back and see what's different (I did have to alter the goal animation on my original one, as it was set off the side of the screen, but that edit didn't cause any issues).

riffnshred, 2020-08-08:
This is fixed in 1.3.0 !!!

closed 2020-08-08 by riffnshred