riffnshred/nhl-led-scoreboard #123: Goal Light Animation Offset

jeepdrvr71, 2020-08-02:
When goal is scored it appears the goal light animation is pushed to the left side of the screen and squished.

cmcurtis33, 2020-08-02:
I'm somewhat new to python and Git/GitHub, but was also experiencing the same issue. Can offer a fix that worked for me.

Within nhl-led-scoreboard/src/renderer folder, edit the main.py file (highly recommend making a backup first via command 'cp main.py main.py.orig').

Scroll down to the function def _draw_goal(self, id, name) [around line 215]. Near the bottom of that function you will find the following line [around line 250]:

  self.matrix.draw_image((0, 0), im,"center")

change it to the following:
  self.matrix.draw_image((self.matrix.width/2, 0), im,"center")

Save and run the app again. Will do some research on how to use Git/GitHub so that this can be part of the next release.

riffnshred, 2020-08-02:
This is fixed in the beta version which will come out Tuesday or Wednesday.

cmcurtis33, 2020-08-02:
@riffnshred thank you!

riffnshred, 2020-08-08:
This is fixed in 1.3.0 !!!

closed 2020-08-08 by riffnshred