riffnshred/nhl-led-scoreboard #124: Errors with new beta

desigi, 2020-08-02:
After having updated to the new beta following @riffnshred notification in discord last night, my board is currently only displaying the clock and then going to a black screen and back to the clock.

I checked the logs and the only errors that I noticed are the following:

`ERROR WHILE RENDERING: 'ScoreboardConfig' object has no attribute 'weather_alert_title'`
and
`ERROR WHILE RENDERING: 'Data' object has no attribute 'wxalert_show_on_clock'`

This is what my config file looks like:
```
{
    "debug": true,
    "live_mode": true,
    "preferences": {
        "time_format": "24h",
        "end_of_day": "9:00",
        "location": "Montreal, QC",
        "live_game_refresh_rate": 10,
        "teams": [
            "Canadiens"
        ],
        "sog_display_frequency": 4
    },
    "goal_animations": {
        "pref_team_only": true
    },
    "states": {
        "off_day": [
            "clock",
            "weather",
            "wxalert",
            "scoreticker",
            "seriesticker",
            "standings",
            "team_summary"
        ],
        "scheduled": [
            "clock",
            "weather",
            "wxalert",
            "scoreticker",
            "seriesticker",
            "standings",
            "team_summary"
        ],
        "intermission": [
            "scoreticker",
            "seriesticker",
            "clock"
        ],
        "post_game": [
            "clock",
            "weather",
            "wxalert",
            "scoreticker",
            "seriesticker",
            "standings",
            "team_summary"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "seriesticker": {
            "preferred_teams_only": false,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "wild_card",
            "divisions": "atlantic",
            "conference": "eastern"
        },
        "clock": {
            "duration": 10,
            "hide_indicator": false
        },
        "covid19": {
            "worldwide_enabled": false,
            "country_enabled": false,
            "country": [
                "USA"
            ],
            "us_state_enabled": false,
            "us_state": [
                "New York"
            ],
            "canada_enabled": false,
            "canada_prov": [
                "Quebec"
            ]
        },
        "weather": {
            "enabled": true,
            "units": "metric",
            "update_freq": 5,
            "view": "full",
            "duration": 30,
            "data_feed": "EC",
            "owm_apikey": "",
            "show_on_clock": true
        },
        "wxalert": {
            "show_alerts": true,
            "alert_feed": "EC",
            "alert_title": false,
            "scroll_alert": true,
            "alert_duration": 10,
            "update_freq": 5,
            "show_on_clock": true
        }
    },
    "sbio": {
        "dimmer": {
            "enabled": false,
            "source": "software",
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": true,
            "bonnet": true,
            "pin": 25,
            "reboot_duration": 2,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 10,
            "poweroff_override_process": "",
            "display_halt": true,
            "state_triggered1": "weather",
            "state_triggered1_process": ""
        }
    }
}
```

I'm on a Pi 3B+ with an Adafruit HAT and with the anti-flicker mod.

riffnshred, 2020-08-02:
You need to re run the nhl_setup and reconfigure since the there are new options added. Check out the annoncement on discord for more details

desigi, 2020-08-02:
@riffnshred I did rerun the nhl_setup and I'm still only getting the clock.

desigi, 2020-08-02:
There appears to be an issue with the alerts board. If I disable it, the rest of the boards appear correctly.

riffnshred, 2020-08-02:
My bad you are right. Thanks for finding this

riffnshred, 2020-08-08:
This was fixed in the official release !!!

closed 2020-08-08 by riffnshred