riffnshred/nhl-led-scoreboard #129: Latest Update Error - "No module named 'lastversion' in UpdateChecker

Tanosh11, 2020-08-06:
Matrix isn't displaying anything after the latest update. Supervisor isn't showing anything, but running the command manually displays:

Traceback (most recent call last):
  File "src/main.py", line 17, in <module>
    from update_checker import UpdateChecker
  File "/home/pi/nhl-led-scoreboard/src/update_checker.py", line 4, in <module>
    from lastversion import lastversion
ModuleNotFoundError: No module named 'lastversion'

Pi 3B+
Using Bonnet
Using Anti-flicker mod
Software version 1.3.0

Can provide config if needed, I reran it after the update to config the new features, but it wasn't working before that either.

riffnshred, 2020-08-06:
For some reason, the install.sh seems to not install it sometimes. If you get that error message, install lastversion manually like so
`sudo pip3 install lastversion`.

Tanosh11, 2020-08-06:
Back in business, thank you sir!

closed 2020-08-06 by Tanosh11

riffnshred, 2020-08-06:
You welcome, Enjoy !!!