riffnshred/nhl-led-scoreboard #130: Board getting exception from  NHL api call that's not handled

falkyre, 2020-08-06:
**Describe the bug**
Seeing on one of my boards, calls to the NHL api will get errors for connection and time out.  Then I'm seeing an exception stating ```NameError: name 'self' is not defined```

**Logs**
```INFO (15:51:50): NHL-LED-SCOREBOARD - v1.3.0 (64x32)
INFO (15:51:50): Now validating config.json.....
INFO (15:51:50): config.json passes validation
Config file config/fonts/fonts_64x32.json not found!
Config file config/layout/layout_64x32.json not found!
INFO (15:51:50): location is: Winnipeg,CA [49.8844, -97.147]
DEBUG (15:51:50): No next game detected for Devils
DEBUG (15:51:50): No next game detected for Rangers
DEBUG (15:51:50): No next game detected for Sabres
DEBUG (15:51:50): No next game detected for Senators
DEBUG (15:51:50): No next game detected for Hurricanes
DEBUG (15:51:50): No next game detected for Red Wings
DEBUG (15:51:50): No next game detected for Ducks
DEBUG (15:51:50): No next game detected for Kings
DEBUG (15:51:50): No next game detected for Sharks
INFO (15:51:51): defaultround number is : 0
INFO (15:51:51): Using software to determine dimming
INFO (15:51:51): /sbin/reboot <-- reboot || poweroff --> /sbin/poweroff
INFO (15:51:51): Checking for dimmer
INFO (15:51:51): Using software to determine dimming
INFO (15:51:51): It is day time
INFO (15:51:51): Refreshing EC current observations weather
INFO (15:51:51): Checking for EC weather alerts
INFO (15:51:51): Checking for new release for falkyre/nhl-led-scoreboard repo installed in /home/dietpi/source/weather
INFO (15:51:52): No new release.
INFO (15:51:53): Rendering...
DEBUG (15:51:53): refresing data
INFO (15:51:54): defaultround number is : 0
INFO (15:51:54): Game Day Wooooo
INFO (15:51:54): Showing Game
INFO (15:51:54): Checking for new day
INFO (15:51:54): It is not a new day
INFO (15:51:54): Game is Scheduled
INFO (15:51:54): Showing Pre-Game
INFO (15:51:56): ['08/06 03:51 PM', '\uf002', 'Partly Cloudy', '28.4C', '32.5C', '45%', '15.6C']
INFO (15:51:56): ['22.0 kph', 'S', '\uf05c', '35.0 kph', '1013.0 hPa', '\uf088', '24.1 km']
INFO (15:51:56): Last Alert: []
INFO (15:51:56): No active EC weather alerts in your area
DEBUG (15:54:26): refresing data
INFO (15:54:26): defaultround number is : 0
INFO (15:54:26): Checking for new day
INFO (15:54:26): It is not a new day
INFO (15:54:26): Game is Scheduled
INFO (15:54:26): Showing Pre-Game 
```
Exception Log
https://pastebin.com/d16kBudc 

**Your config file**
https://pastebin.com/AUyndqQb

**Hardware you are using**
- Raspberry pi model 3A+
- The Matrix driver (Adafruit Bonnet or Adafruit HAT) Bonnet
- Anti-flicker mod or not?  yes

**Software**
- Which version of the software are you on? 1.3.0

harmgsn, 2020-08-07:
the exception logs show "temporary error in name resolution" which indicates a network issue -- IE:  DNS wasn't able to resolve the URLs for the APIs and thus crashed....

falkyre, 2020-08-07:
Yup. The code to handle that exception is missing.  So when it comes up the code fails instead of trying again.

closed 2020-08-18 by falkyre