riffnshred/nhl-led-scoreboard #131: Traceback in Error log

harmgsn, 2020-08-07:
Traceback (most recent call last):
  File "src/main.py", line 117, in <module>
    run()
  File "src/main.py", line 112, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 45, in render
    self.__render_offday()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 69, in __render_offday
    self.data.refresh_data()
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 483, in refresh_data
    self.refresh_games()
  File "/home/pi/nhl-led-scoreboard/src/data/data.py", line 241, in refresh_games
    self.games = nhl_api.day(self.year, self.month, self.day)
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/__init__.py", line 18, in day
    return [nhl_api.game.GameScoreboard(data[x]) for x in data]
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/__init__.py", line 18, in <listcomp>
    return [nhl_api.game.GameScoreboard(data[x]) for x in data]
TypeError: 'Response' object is not subscriptable

(Config file attached -- key edited out)
[config.json.txt](https://github.com/riffnshred/nhl-led-scoreboard/files/5038764/config.json.txt)

riffnshred, 2020-08-08:
Your config looks good. This seems to be caused because the request to API failed somehow. Is it an intermittent issue or it constantly happens. Hows the wifi where you scoreboard situated?

topher920, 2020-08-20:
I have a similar issue. Everything was working fine until I updated.

`INFO (22:30:33): Showing Post-Game
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 343, in _make_request    self._validate_conn(conn)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 841, in _validate_conn
    conn.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 344, in connect
    ssl_context=context)
  File "/usr/lib/python3/dist-packages/urllib3/util/ssl_.py", line 344, in ssl_wrap_socket
    return context.wrap_socket(sock, server_hostname=server_hostname)
  File "/usr/lib/python3.7/ssl.py", line 412, in wrap_socket
    session=session
  File "/usr/lib/python3.7/ssl.py", line 853, in _create
    self.do_handshake()
  File "/usr/lib/python3.7/ssl.py", line 1117, in do_handshake
    self._sslobj.do_handshake()
socket.timeout: _ssl.c:1039: The handshake operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 367, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/usr/lib/python3/dist-packages/six.py", line 693, in reraise
    raise value
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 346, in _make_request    self._raise_timeout(err=e, url=url, timeout_value=conn.timeout)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 306, in _raise_timeout
    raise ReadTimeoutError(self, url, "Read timed out. (read timeout=%s)" % timeout_value)
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='records.nhl.com', port=443): Read timed out. (read timeout=5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/data.py", line 97, in get_series_record
    data = requests.get(SERIES_RECORD.format(seriesCode, season), timeout=REQUEST_TIMEOUT)
  File "/usr/local/lib/python3.7/dist-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.7/dist-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/requests/adapters.py", line 529, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='records.nhl.com', port=443): Read timed out. (read timeout=5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "src/main.py", line 117, in <module>
    run()
  File "src/main.py", line 112, in run
    MainRenderer(matrix, data, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 53, in render
    self.__render_game_day()
  File "/home/pi/nhl-led-scoreboard/src/renderer/main.py", line 141, in __render_game_day
    self.boards._post_game(self.data, self.matrix,self.sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 148, in _post_game
    board(data, matrix,sleepEvent)
  File "/home/pi/nhl-led-scoreboard/src/boards/boards.py", line 164, in seriesticker
    Seriesticker(data, matrix, sleepEvent).render()
  File "/home/pi/nhl-led-scoreboard/src/boards/seriesticker.py", line 37, in render
    series = Series(s,self.data)
  File "/home/pi/nhl-led-scoreboard/src/data/playoffs.py", line 55, in __init__
    self.games = nhl_api.series_game_record(self.series_code, data.playoffs.season)
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/__init__.py", line 52, in series_game_record
    return nhl_api.info.series_record(seriesCode, season)
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/info.py", line 118, in series_record
    data = data = nhl_api.data.get_series_record(seriesCode, season)
  File "/home/pi/nhl-led-scoreboard/src/nhl_api/data.py", line 100, in get_series_record
    raise ValueError(e)
ValueError: HTTPSConnectionPool(host='records.nhl.com', port=443): Read timed out. (read timeout=5)
`

riffnshred, 2021-01-03:
I believe this is fixed in the recent update. if not, I recommend deleting the nhl-led-scoreboard folder and start over at the software installation in the documentation.

closed 2021-01-04 by riffnshred