riffnshred/nhl-led-scoreboard #134: 1.3.1 Upgrade killed my display

meClaudius, 2020-08-09:
**Describe the bug**
After completing the upgrade from 1.1.6 to 1.3.1 which took about 30 minutes. The only warning I noted (could I have missed one?) Was that it would not upgrade led-rgb-matrix as it could not determine dependancies. I figured that was fine.
Since I auto start it I did not see logs when it actually started.  It did start up and was running but the LED matrix was blank.
So I kill the program and started it manually from terminal from root and got these messages 

**Logs**
sudo /usr/bin/python3 /home/pi/nhl-led-scoreboard/src/main.py --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60 --led-slowdown-gpio=2
Traceback (most recent call last):
  File "/home/pi/nhl-led-scoreboard/src/main.py", line 3, in <module>
    from data.scoreboard_config import ScoreboardConfig
  File "/home/pi/nhl-led-scoreboard/src/data/scoreboard_config.py", line 5, in <module>
    from nhl_setup.validate_json import validateConf
  File "/home/pi/nhl-led-scoreboard/src/nhl_setup/validate_json.py", line 1, in <module>
    import fastjsonschema
ModuleNotFoundError: No module named 'fastjsonschema'

**Your config file**
A screen shot of your config.json file can be really useful to help you. 
{
    "debug": false,
    "live_mode": true,
    "preferences": {
        "time_format": "12h",
        "end_of_day": "8:00",
        "live_game_refresh_rate": 15,
        "teams": [
            "Flyers"
        ],
        "sog_display_frequency": 4
    },
    "goal_animations": {
        "pref_team_only": true
    },
    "states": {
        "off_day": [
            "clock",
            "scoreticker",
            "standings",
            "team_summary"
        ],
        "scheduled": [
            "scoreticker",
            "team_summary"
        ],
        "intermission": [
            "scoreticker",
            "team_summary"
        ],
        "post_game": [
            "clock",
            "standings"
        ]
    },
    "boards": {
        "scoreticker": {
            "preferred_teams_only": true,
            "rotation_rate": 5
        },
        "standings": {
            "preferred_standings_only": true,
            "standing_type": "conference",
            "divisions": "metropolitan",
            "conference": "eastern"
        },
        "clock": {
            "duration": 8,
            "hide_indicator": true
        },
        "covid19": {
            "worldwide_enabled": false,
            "country_enabled": false,
            "us_state_enabled": false,
            "canada_enabled": false
        }
    },
    "sbio": {
        "dimmer": {
            "enabled": false,
            "source": "software",
            "frequency": 5,
            "light_level_lux": 400,
            "mode": "always",
            "sunset_brightness": 5,
            "sunrise_brightness": 40
        },
        "pushbutton": {
            "enabled": true,
            "bonnet": false,
            "pin": 25,
            "reboot_duration": 3,
            "reboot_override_process": "",
            "display_reboot": true,
            "poweroff_duration": 8,
            "poweroff_override_process": "Y",
            "display_halt": true,
            "state_triggered1": "clock",
            "state_triggered1_process": "Y"
        }
    }
}

**Hardware you are using**
- Raspberry pi model 3B+
- The Matrix driver (Adafruit Bonnet or Adafruit HAT) Adafruit HAT 
- Anti-flicker mod or not? Yes pam jumper

**Software**
- Which version of the software are you on? Upgrading 1.1.6 to 1.3.1

riffnshred, 2020-08-09:
Sorry for that m8, This is due to the number of new features and in consequence, new modules needed to run it. 
two things you can do:

1. run these commands
`git reset --hard`
`git pull`
`sudo chmod +x scripts/install.sh`
`sudo ./scripts/install.sh`

This will update everything in software. Then run the nhl_setup to reconfig because there are new options in there.

if the does not work...

2. Delete de software and clone again.
`cd` to get to the root of your user folder
run :
`sudo rm -R nhl-led-scoreboard`

than run the steps to install the software that are in the Documentation [here](https://github.com/riffnshred/nhl-led-scoreboard#installing-the-nhl-scoreboard-software): 

Let me know if this does not fix it.

meClaudius, 2020-08-09:
Thanks for the help.  The hard reset and pull did the trick.  I used the setup and then edited. A couple of the drop down pickers challenged me but I edited it afterwards ...Thank you

riffnshred, 2020-08-09:
Sweet! Enjoy

closed 2020-08-09 by riffnshred