riffnshred/nhl-led-scoreboard #135: did the update to 1.3.2, but scoreboard is still giving green bar at top. Seems that issue could be that VERSION file is still having 1.3.1. Not a big deal, but wanted to mention

cmcurtis33, 2020-08-09:
**Describe the bug**
A clear and concise description of what the bug is.

**Logs**
Copy and past any error or messages that you can get. 

**Your config file**
A screen shot of your config.json file can be really useful to help you. 

**Hardware you are using**
- Raspberry pi model
- The Matrix driver (Adafruit Bonnet or Adafruit HAT)
- Anti-flicker mod or not? 

**Software**
- Which version of the software are you on?

riffnshred, 2020-08-09:
sorry about that mate, I just bumped the version in VERSION file. let me know if that didn fix it

cmcurtis33, 2020-08-09:
Yes that worked.. glad it was an easy fix. Thanks again for all that you are doing to make the hockey season more enjoyable -- and have a cool toy to keep track of scores while I'm working!

riffnshred, 2020-08-09:
Thanks man , glad you enjoy !!! Fun fact I made this thing to also keep track of scores while I'm working !!!

Cheers !!!

closed 2020-08-09 by riffnshred