riffnshred/nhl-led-scoreboard #136: Auxiliary output for light flash

jjcustomz, 2020-08-10:
Would it possible to use GPIO pin(s) to control an external light (LED strip, rotating beacon, etc)? I'm not sure if that's possible using the header already. Ideally an output to control an external relay would be cool, as you could then get creative.

I have been looking to install [this nhl goal light](https://github.com/arim215/nhl_goal_light) and wondering if instead of using a different Pi, it could be integrated into this amazing project.

Thanks for all your hard work, I have been using this since one of the first versions, and it just keeps getting better.

riffnshred, 2020-08-10:
Of course, it can be done. If you look at the SBIO documentation [here](https://github.com/riffnshred/nhl-led-scoreboard/blob/master/src/sbio/SBIO.md) you can see which pins are available. The current version does not support it but some guys like @falkyre forked and modified their version of the software to trigger all kinds of stuff. I suggest you to come join the discord channel, there is a bunch of guys doing modified versions and like triggering IoT stuff, lights and sounds.

riffnshred, 2020-08-10:
Since more and more people are looking to do stuff like that, I'll add this to the list of enhancements.

jjcustomz, 2020-08-10:
Awesome, thank you very much for the information. I will definitely check it out!

falkyre, 2020-08-10:
It's doable with a catch.  The bonnet and the HAT use a heck of a lot of GPIO pins.  However, that being said you should be able to add a 5V relay to the Pi and use the open GPIO pins pretty easy.  I've added motion sensor and also a local temperature sensor along with the pushbutton and there's enough pins there to do it.   In fact, I've looked at that same github that you mention and thought about adding that but I have a BudlLight Red Light I want to hack first.

jjcustomz, 2020-08-11:
I like the idea of the Bud Red Light, but can't justify the price. This project has still cost less, and already does more. By the sounds of it, can do even more. I'm not familiar with Python, my background is web development and have used most Javascript, with some c++ in school, but the last ~12 years I have been working with car electronics and still try to tinker. Would be amazing if a feature like this could be implemented into the project, but I may also have to try some other things if I can find the time too. Home Assistant is also on my list as I see you also use.