riffnshred/nhl-led-scoreboard #137: Poller not updating to game day

cheechbg, 2020-08-13:
I really love the work that's gone into this scoreboard, it's come a LONG way since the MLB port.  I seem to be having an issue with the API calls and updating the board to game day.  I have the Blue Jackets set as the primary team, yet the board is still returning this:

DEBUG (10:55:03): PING !!! Render off day

The NHL API (https://statsapi.web.nhl.com/api/v1/schedule) is returning yesterday's scores (I thought it was supposed to be today's?)

Adding the [?date=2020-08-13] to the end of the URL does return today's scheduled games.  Is the Python script just thinking the first URL without the addition is supposed to come back with today instead of yesterday?

Again, thanks a lot!

Jeff

riffnshred, 2020-08-13:
Thanks man. Make sure the "end of day" option is set between 00:01 and Noon. This option bear a confusing name and will change on the next update. Its to deal with the NHL API showing the previous day schedule up to Noon.

Cheers

cheechbg, 2020-08-13:
Tried changing the end-of-day to 08:00 per the default config.json, no change.  Debug info:

DEBUG (12:23:14): PING !!! Render off day
INFO (12:23:14): Checking for new day
INFO (12:23:14): It is not a new day

Checking the NHL API feed, I see they changed their default schedule to today instead of yesterday.

riffnshred, 2020-08-13:
check if your "Live Mode" is set to true. When its set to false, it stay in the Off day state

riffnshred, 2020-08-13:
otherwise, post your config here. Might be something weird in there

cheechbg, 2020-08-13:
Bingo, that was it, live mode was off.  Thanks!

closed 2020-08-13 by cheechbg

riffnshred, 2020-08-13:
Enjoy bud !!!