riffnshred/nhl-led-scoreboard #138: can't open file 'runtext.py'

nmsidey, 2020-08-14:
**Describe the bug**
Cant run sample. I downloaded software last night. Was able to run sample text but not full program. Installed AntiFlicker Mod this morning and deleted nhl-led-scoreboard directory. Reinstalled directory, cannot run sample or full program

**Logs**
pi@raspberrypi:~/nhl-led-scoreboard $ sudo python3 runtext.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat-pwm --led-brightness=60
python3: can't open file 'runtext.py': [Errno 2] No such file or directory

**Hardware you are using**
I'm using all recommended hardware. Pi Zero WH, Adafrouit Hat, Bonnet and LED Board and the Anti Flicker Mod

**Software**
- Which version of the software are you on?
My software is up to date with the latest build.

riffnshred, 2020-08-14:
check out the documentation, you forget one line before running the runtext.py ;)

https://github.com/riffnshred/nhl-led-scoreboard#testing-and-optimization

nmsidey, 2020-08-14:
pi@raspberrypi:~/nhl-led-scoreboard $ cd nhl-led-scoreboard/submodules/matrix/bindings/python/samples
-bash: cd: nhl-led-scoreboard/submodules/matrix/bindings/python/samples: No such file or directory
pi@raspberrypi:~/nhl-led-scoreboard $ 

Am I missing something else?

nmsidey, 2020-08-14:
I will gladly pay someone to FaceTime/take control of my Pi to help me resolve this. I have a watch party tonight and Id like to get it running before hand.

riffnshred, 2020-08-14:
get on the discord in the help section I'll help you out from there

closed 2020-08-15 by nmsidey