riffnshred/nhl-led-scoreboard #139: Option to have display turn off at night

beamo11, 2020-08-15:
Would it be possible to add the ability to automatically turn off and on the display at a certain time so it's not running all night?

riffnshred, 2020-08-15:
For sure using supervisor and supervisorcli. If someone wants to look into it during the playoff I can accept a pull request. Otherwise, it will probably be added for the next season.

falkyre, 2020-08-15:
The other option is to set your brightness to 0 at night if you are just looking at not having display on and the code still running. I'll be adding some more functionality to the dimmer in next few weeks so I could add an off time and on time as well

beamo11, 2020-08-16:
I tried setting the brightness to 0 but that doesn't actually turn it off it just becomes very dim. It may require pasuing the rendering but I'm not sure.

riffnshred, 2020-08-16:
I blocked the brightness to 1 in the code because if it's running, you should be able to see it. I know there will be this guy who will think the panel is broken when they see the scoreboard is running on supervisor but nothing shows up and the reason why is because they set their brightness to zero.

Using this script as example and some research on google, it looks pretty straight forward to do. https://github.com/necromancyr/swap-led-scoreboard

@falkyre that can be a temporary alternative for now but once a more solid solution is done I'll block the minimum to 1 again

falkyre, 2020-08-16:
Sure.  I'll still look into adding an "off" function with the stuff I'm doing with the dimmer.   This will keep the board running just shut off display. The supervisor way will shut the boards down and stop the code running.  Either option will work. So if I add the dimmer option will you allow brightness 0 then?

riffnshred, 2020-08-16:
My issue with dimming the board to zero is it's a "hack" instead of making a feature.  it opens doors to confusion and mistakes in the long run and it should definitely be part of the system that controls the board instead. I get where you come from and it's the fastest and easier way, and I think temporarily this would do the job, but I wouldn't advertise it at a feature and more like a trick or a hack. 

At a later date if no one makes this feature I'll get on it for sure because I do shut it off manually half the time using my phone and I like the idea of not even thinking about it.

falkyre, 2020-08-16:
The easiest way to actually do is is two Cron jobs. One at the time you want the board off, do a supervisorctl start scoreboard. And a second when you want it on, supervisorctl start scoreboard.  That's standard Linux and not part of the code. I was looking at inside the code but I'll leave it. It'd be better just to use Cron to do it.

riffnshred, 2020-08-16:
I see what you mean with the screen saver and I like that Idea. Just make sure it's in the debug. 

What I mean by hack is using a feature that was not meant for doing this and just adding a couple of lines to accomplish something else. Like using a wrench as a hammer lol. I think of the dimmer as something specific to control the brightness at a specific time of the day to ease the view, but with a screen saver, I see it more like a different mode that the board gets on. 

Also an idea for that, Let's say for some reason we want to get out of the screen saver, the button could come handy.

falkyre, 2020-08-16:
I'll make a screen saver class seperate from dimmer.

falkyre, 2020-08-26:
> I tried setting the brightness to 0 but that doesn't actually turn it off it just becomes very dim. It may require pasuing the rendering but I'm not sure.

Just starting the work on the screensaver functionality.  @beamo11 Where did you set the brightness to 0?  @riffnshred I only see one place in all of the code that has brightness set to 1 but the dimmer "should" override that.  I don't see anywhere that you are forcing the value if it's 0 to be 1.

beamo11, 2020-08-26:
I only set it in the config. I'm not sure where else it would be.

falkyre, 2020-08-26:
Where in the config? The dimmer section? You can set it on the command line as well.

riffnshred, 2020-08-26:
@falkyre I think thats it. Since that how we set the brightness (other than the dimmer in config) thats where I set the limit to 1. Dont remember by heart tho.

jeepdrvr71, 2020-08-29:
Could we have the display not look at the nhl api or weather api during dimming hours. This would reduce read/writes to the sd card and extend their life. 

Has anyone tried using an SSD with this project? just curious

riffnshred, 2020-08-29:
This is where I was going at first with this idea. It can be done for sure but I won't have that for the current playoff. Your best option is to set a CRON job to start and stop the software at whatever time you want.

As for SSD not that I know of, but I know it is doable.

falkyre, 2020-08-30:
I'll see what I can add for pausing calls to API. I've switched everything weather and dimmer related to a scheduler that can be paused so for sure weather stuff can be paused.  SSD is doable, I have one of my rpi4s booting off SSD. But for this project I use dietpi which does all logging to ramlog. Saves the sd card.

falkyre, 2020-08-30:
You could set up your own Ram tmpfs and have the supervisor logs and the scoreboard logs go there instead of to the sd card. Here's an article on doing it.  https://afterhourscoding.wordpress.com/2017/09/06/using-supervisord-and-tmpfs-ramdisks-on-a-raspberry-pi/

falkyre, 2020-09-30:
This is in PR #144

closed 2021-01-03 by riffnshred